karawin/Ka-Radio #1: auto resume

zajaklis, 2016-06-23:
At fist i wanted to thank you for you hard work and time spent on this project.

This not an issue, but more like feature request.
It would be nice that device on startup starts automatically playing radio station which was running before shutdown.
At fist i thought that its configurable on webui with "AutoPlay" checkbox, but it did not do anything.
What AutoPlay checkbox does?

karawin, 2016-06-23:
Hello,

Thanks,

Yes, the autoPlay is only after the first play button pressed. After 
that, any other selection is launched automatically.
I forgot to launch the current station. I will work on it soon.

Regards

JP

closed 2016-07-29 by karawin

zajaklis, 2016-09-07:
hi again, did you implemented auto resume feature?
i have not notice that in changelog.

karawin, 2016-09-08:
Partially.
The problem is that the knowing is not in the esp but in the browser.
I have to add it on the esp eprom.... Soon.

karawin, 2016-09-23:
autostart: test these bin....
I think it will works now.
thanks

zajaklis, 2016-09-24:
which bins include autostart feature?

karawin, 2016-09-27:
Both.

Two bin, one for each bank.

Try just the 1.

zajaklis, 2016-09-27:
took from http://karadio.karawin.fr/ and autoplay still not working after applying power.
Idea was to interact  with ka-radio without webui. At the moment after applying power im counting till 10 and pressing play button.

karawin, 2016-09-27:
Normal, it is not delivered, the bin was a preview .

karawin, 2016-09-29:
The 1.0.9 is released and working ;-)

zajaklis, 2016-10-01:
Great, just tested, all works!
Thank you