karawin/Ka-Radio #2: cannot change IP

yo2ldk, 2016-07-16:
Hi,

  this is a very nice and well made it project, Thank you!
I do something wrong maybe, because when I try to put AP name, pswd , a new IP and gateway, this is not happening , it comes back on 192.168.4.1 and I have no connection to internet for ESP module.
on router I have ...1.1, try it also with original class ...8.1 and with ...0.1 , all the same.
I not have memory IC (until I get it),  but it work also without it, no? and when ESP have no power, I need to rewrite all settings in this case?
 sorry for all questions, but I am more on hardware and a beginner in codding .

karawin, 2016-07-16:
Hi

Firstly check if you have a 4Mb esp8266 (ie 32mbits)

>
> Hi,
>
> this is a very nice and well made it project, Thank you!
> I do something wrong maybe, because when I try to put AP name, pswd , 
> a new IP and gateway, this is not happening , it comes back on 
> 192.168.4.1 and I have no connection to internet for ESP module.
May be a wrong AP or password?
Before all, make a "Clear stations" and after change the wifi setting: 
No dhcp, an ip address in 192.168.1.x a mask of 255.255.255.0 and the ip 
of your router.

> on router I have ...1.1, try it also with original class ...8.1 and
> with ...0.1 , all the same.
> I not have memory IC (until I get it), but it work also without it,
> no? and when ESP have no power, [...]

Yes it works without the external ram.
Check for a 4MB esp8266 if the stations are not maintened, it may be the 
problem.
If it is the wifi setting not maintened, it is because the internet AP 
access is wrong.

yo2ldk, 2016-07-16:
Thank you for fast reply
 I try for 3 days already and no conclusion..
yes my esp board have 4Mb is wemos esp d1 mini, but I try also with NodeMcu v.3 the same thing
 firmware is well flashed ,router is ok, password all good but not connect
web interface it is and work ok, full level of wifi, connections is made it conf. pdf. file; i'm stuck...  :)

karawin, 2016-07-16:
Try the uart interface (documented in 
\a-Radio\ESP8266-Firmware\documents\uartinterface.txt):

sys.erase
wifi.list
wifi.con("ssid","password")
wifi.status

Bye

JP

yo2ldk, 2016-07-16:
Merci beaucoup !

karawin, 2016-07-16:
Check also your power. It is very sensitive. A good external power 
supply is needed.

yo2ldk, 2016-07-16:
I use power bank Xiaomi 16000 mA

karawin, 2016-07-16:
115200b/s ?
try http://www.compuphase.com/software_termite.htm
Press enter key, you must see
#INFO:""#

If not, the flash is not good.

yo2ldk, 2016-07-16:
sorry to bother you with my problem;
I use tera term, also realterm, now on termite when push enter nothing appear. all on 9600
testing also with 115200
web page look ok, I reflashing, last solutions here.

yo2ldk, 2016-07-16:
for sure i wrong somthing..
after reflashing two modules, with serial connection I have no response from any.  at all !
 I flash as follow:

boot_v1.5 at 0x00000
user1.4096.new.4.bin at 0x1000 ,
user2.4096.new.4.bin at 0X81000
blank.bin at 0x7e000 
blank.bin at 0x3fe000

it is ok that?  because I don't know what to do more..

karawin, 2016-07-16:
The complete flash is
[image]

With what tool do you flash it?
Under windows or linux? 
Do you have checked all wires? Try without the vs1053 in case of doubt.

karawin, 2016-07-29:
no news?

closed 2016-07-29 by karawin

yo2ldk, 2016-07-29:
Jean-Pierre, 
Thank you for all your help, especially latest, with photo !
 in the first time, I write firmware exactly like you except latest two file .  
 Now i write all files, on four (4) different modules. 
 on all when I try to access with terminal (5 different software) I get nothing or strange char.
not what should be. 
I'm not so stupid, i load firmware in many and different equipment's but with that this is like I fell.. 
 so , I take a break, maybe in this weekend I made latest try, even if I need this Radio.
 my best regards,

                 alex.

karawin, 2016-07-29:
[image]