karawin/Ka-Radio #4: Bug selected stations  autoplay

lyovav, 2016-10-06:
Sometimes when I chose the desired station, and I pressed the play - off the power and turned it back to station plays 1 second and stops playing
```
connected with 47501, channel 3
dhcp client start...
ip:192.168.100.246,mask:255.255.255.0,gw:192.168.100.1
autostart: playing:1, currentstation:2
I2S Speed: 0
##CLI.STOPPED#
##CLI.URLSET#: online-hitfm.tavrmedia.ua
##CLI.PATHSET#: /HitFM
##CLI.PORTSET#: 8000
##CLI.ICY0#: Hit FM
##CLI.ICY3#: http://www.hitfm.ua
##CLI.ICY4#: Only Hits!
##CLI.ICY5#: 128
##CLI.ICY6#: Only Hits!
##CLI.ICY7#:  ice-samplerate=44100;ice-bitrate=128;ice-channels=2
##CLI.PLAYING#
##CLI.STOPPED#
```

In most cases operate normally Autoplay's

```
connected with 47501, channel 3
dhcp client start...
ip:192.168.100.246,mask:255.255.255.0,gw:192.168.100.1
autostart: playing:1, currentstation:2
I2S Speed: 0
##CLI.STOPPED#
##CLI.URLSET#: online-hitfm.tavrmedia.ua
##CLI.PATHSET#: /HitFM
##CLI.PORTSET#: 8000
##CLI.ICY0#: Hit FM
##CLI.ICY3#: http://www.hitfm.ua
##CLI.ICY4#: Only Hits!
##CLI.ICY5#: 128
##CLI.ICY6#: Only Hits!
##CLI.ICY7#:  ice-samplerate=44100;ice-bitrate=128;ice-channels=2
##CLI.PLAYING#
##CLI.META#: LOBODA - К черту любовь
```

karawin, 2016-10-06:
Interesting. I will investigate, thank you

lyovav, 2016-10-10:
```
=> extraram state: 0 0xff ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea7c0, len: 7904
bss   : 0x3ffea7c0 ~ 0x3fff1620, len: 28256
heap  : 0x3fff1620 ~ 0x40000000, len: 59872
Heap size: 32088
t0 task: 3fff8458
t1 task: 3fff85f0
t4 task: 3fff8a18
t3 task: 3fff9038
t2 task: 3fff9d60
mode : sta(5c:cf:7f:23:d6:53)
add if0
I2S Speed: 0
SCI_Mode (0x4800) = 0x802
SCI_Status (0x48) = 0x48
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0x6000
UART READY TO READ
 Station Ip: 192.168.100.247
DHCP: 0x1
Device: Ip: 192.168.100.247

I: 0 status: 0

In I: 0 status: 1
scandone
state: 0 -> 2 (b0)

In I: 1 status: 1
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
pm open phy_2,type:2 0 0
cnt

connected with 47501, channel 3
dhcp client start...
ip:192.168.100.247,mask:255.255.255.0,gw:192.168.100.1
autostart: playing:1, currentstation:0
I2S Speed: 0
##CLI.STOPPED#
##CLI.URLSET#: ua.uar.net
##CLI.PATHSET#: /galychyna
##CLI.PORTSET#: 8000
##CLI.ICY0#: Galychyna FM
##CLI.ICY3#: http://www.galychyna.fm
##CLI.ICY4#: galychyna
##CLI.ICY5#: 256
##CLI.ICY6#: This is only a trial
##CLI.PLAYING#
##CLI.STOPPED#

```

The new release(V1.0.10) works autostart almost perfect but still with 10 power outages without the inclusion of one autostart(
I do not know what it's done...
Thank you very much for giving the new release.

Thank you with best regards Vladimir.

karawin, 2016-10-10:
I tried about 30 autostart with your station with success.
The only reason in the code is a time out on the data reception.
Could you test with another AP?

lyovav, 2016-10-11:
After reading your advice, I tested on another chip ESP8266 and repeat the same problem with autostart failed. Most likely the problem the chip ESP8266.
Thank you very much for your cooperation.
Best regards Vladimir.

karawin, 2016-10-11:
I don't think that it is a problem of ESP8266.
I said to try another network access point or better another internet 
access.
Thanks to you too. Users are the best testers.

lyovav, 2016-10-11:
I have [https://routerboard.com/rb751u-2hnd] AP and 100Mb LAN Internet access [http://www.speedtest.net/my-result/5706090391] Therefore, I have no doubt in my AP Network.
the problem was resolved by replacing another chіp
[https://ru.aliexpress.com/item/New-Wireless-module-CH340-NodeMcu-V3-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-We/32565317233.html?spm=2114.13010708.0.0.ewVMNO&detailNewVersion=&categoryId=400103]

Best regards Vladimir.

zajaklis, 2016-10-25:
problem was caused because of missing pulldown for analog input. otherwise keypad must be connected.

lyovav, 2016-10-25:
a minimum of parts and without control board
[prototype](https://1drv.ms/i/s!Aq_8ScOZC7u_l37tdNEGBa4C95LH)
works perfect...

karawin, 2016-10-25:
Great
Thanks for the feedback.
Enjoy.

closed 2016-10-25 by karawin