karawin/Ka-Radio #5: no connection to AP

acocalypso, 2016-10-17:
Im having a hard time to get a connection. Sometimes it will connect to my WLAN but most time, the blue led (nodemcu v3) blinks a few times and then nothing happens. After restarting the esp **many** times it will reconnect to my wlan.

karawin, 2016-10-17:
Hi,
This part is the espressif code.
Try to be closer of the AP?
Connect a terminal to see the log.

v222008, 2016-10-18:
Hi. Developer fee (nodemcu v3)  is very demanding to the supply voltage, any voltage drop leads to restart or freezes chip.

karawin, 2016-10-18:
Hi v222008
Right, but i made 4 protos without problem. The advantage of the v3 is that the usb 5v is available on pin VU (3). In case of connection with an usb port too weak, you can use hub with external power
Or a low cost https://fr.aliexpress.com/item/Best-price-30pcs-lot-MB102-Solderless-Breadboard-Power-Supply-Module-3-3V-or-5V/32351807367.html?spm=2114.13010608.0.0.VPHfoW

acocalypso, 2016-10-18:
If its demanding to the suppy voltage, im going to test it again with a wallcharger.
Currently i was using my pc / notebook to power it on. 
Ill let you know if this will solve my issue.

v222008, 2016-10-18:
Not the fact that USB cable or contacts in the micro yusb reliable. Try another USB cable or connect to a power supply of 5 volts.
But in the ideal case, connect the port to monitor and see the log...

karawin, 2016-11-10:
No news, i close it

closed 2016-11-10 by karawin