karawin/Ka-Radio #6: Suggestion - saved volume per station

QuitButton, 2016-10-20:
This is shaping up nicely to be a great internet radio, there's a couple of suggestions though I think would help, the main one being saving the volume (or more accurately, a volume offset) per station. Some of the stations are extremely quiet and some very loud, it would be nice to be able to chance station without fear of having my ears blown out ;)

v222008, 2016-10-21:
Actual situation at the moment, who listens to the radio stations on headphones. Already several times found himself in such a situation. While alive and well)

arlaor, 2016-12-21:
Hi @karawin 
I really like your project, because it allows me to update my old sound system.
I am waiting for the parties to be able to do it.
I do not know if you have already implemented it in your project, but considering the volume, it is possible to save custom equalizer parameters in a particular station ?, it happens the same as with volume, some handle too many bass sounds, other too many treble .

karawin, 2016-12-21:
Thanks.
This feature seems simple to implement, but wait:
- the list of stations  is saved in eprom. Not so easy to add another set of data,
- the list can be saved in a file of the pc:, so the mechanism must be modified (saving and restoring)
- this modification must be compatible with the initial format,
- It is ok for the volume with a delta / current volume, but for bass and treble and cutting frequency?
- and finally, I was operated of a cataract Monday and it is a pity to read my screen ;-)

May in one month or two if all is ok with new glasses ,-)
Thanks for your interrest.

arlaor, 2016-12-21:
Hi @karawin 
The equalizer parameters would be stored in a list of equalization parameters and would be called when playing the station to which the custom equalization was assigned.
I do not know if this is possible

I hope your surgery improves healthily. Take care.
We need you with very good health

karawin, 2016-12-21:
It's good to be supported. To later when I would be in shape.
My hobby is to build the best webradio on esp8266.

arlaor, 2016-12-24:
Hi @karawin 
I was reading the comments of the page of your project [hackaday](http://forum.xda-developers.com/galaxy-s4-mini/development/rom-cyanogenmod-14-0-s4-mini-duos-t3477343/page44) and I have given, that I have requested two modules wrong LOL. I should have purchased the VS1053 and purchased the VS1003. I do not know what I was thinking of. I had selected stations that broadcast in AAC format, but now I can not hear them.

karawin, 2017-01-10:
To answer to arlaor, yes, the vs1053 support aac audio.
The new 1.1.4 release has an offset volume per station and many more.
Hope you all like it.
Yes, only volume offset because of the limited space on eeprom.
I close this request.
If you have some other ideas, don't hesitate.
Enjoy.

closed 2017-01-10 by karawin