karawin/Ka-Radio #9: error after flashing

xCite1986, 2016-11-21:
i connect my nodeMCU 1.0 connected with VS1053 shown as in the instruction

after uploading the FW...

=> extraram state: 0 0xff ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea9a8, len: 8392
bss   : 0x3ffea9a8 ~ 0x3fff1850, len: 28328
heap  : 0x3fff1850 ~ 0x40000000, len: 59312
Heap size: 31528
t0 task: 3fff8688
t1 task: 3fff8820
t4 task: 3fff8c48
t3 task: 3fff9268
t2 task: 3fff9f90
mode : sta(5c:cf:7f:89:2b:92)
add if0

then it stops. anyone know whats wrong??

karawin, 2016-11-21:
If you have a terminal on uart and responding, try a
sys.erase		
command.

xCite1986, 2016-11-21:
yes, only 2x short after pressing reset button.
nothoing happens by the command "sys.erase" with arduino seial monitor :(

karawin, 2016-11-21:
Try to flash a blank.bin file at 0x3E0000 and 0x3F0000
If it still doesn't work, I don't know what to do next, apart change the esp itself.

xCite1986, 2016-11-21:
thanks, i'm flashing the complete esp with 
http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

now it blinks fast... but i could not find the AP "WifiWebRadio"

xCite1986, 2016-11-21:
first many thanks for your help and your fast reply!!

it was very curious... 
after getting only hieroglyphs with my nodeMCU1.0 i get a "new" nodeMCU 0.9.. 

first flashing 0x000000, 0x100000, 0x200000 and 0x300000 with blank 1M-File,

and after uploading files from http://karadio.karawin.fr/ ...
http://www.youscreen.de/xiphpqysh66.jpg
...getting the same output:

sdlÜ<là<lä#|ìsbcò'oß$g'ãäcpìÇds$sdpò'àdcoã|llcÄò'oï$l`gnd`osÛnãì$`pûgàsÄãàcoâ<ìÇbóogçl`o'l`o;Ç'd`8ûgàrÜãàìdcnã|$Ç{#Äóngï$`god`gsÛoûd$`rÛgóll`{loû'oã{no'Äd#{{$älìÛlllrlìddd'ãlldläÛlgà'än?ÄÄddà#ädlp##;äd`olÇxìl`cd`$Û'gÛãnl|cll s$sìbÄã#$b#`;Çx#ìónîogãlò8dlld$d`güÙ
=> extraram state: 0 0xff ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea9a8, len: 8392
bss   : 0x3ffea9a8 ~ 0x3fff1850, len: 28328
heap  : 0x3fff1850 ~ 0x40000000, len: 59312
Heap size: 31528
t0 task: 3fff8688
t1 task: 3fff8820
t4 task: 3fff8c48
t3 task: 3fff9268
t2 task: 3fff9f90
mode : sta(5c:cf:7f:03:96:a5)
add if0

is there something wrong with my connection to VS1053-Board?

5V to 5V
5V to 5V
MISO - D6
SCK - D5
XRST - D3
XDCS - D1
XS - D8
DREQ - D2
MOSI - D7
DGND - GND

karawin, 2016-11-21:
Correct.
Dont forget to wire A0 to Gnd
And +5v to VU on nodeMcu, not to the Vin of the nodemcu.

Tho output is correct until add if0.
It seems that the spi bus is blocking.

karawin, 2016-11-21:
Try to remove the vs153 board. It starts without it (but no sound obviously)

v222008, 2016-11-22:
Take advantage of all the functionality of the project is possible without connecting vs153 board like connect to AP, browsing web pages, make a settings  of project...
May be you wrong ESP8266 flashing...

xCite1986, 2016-11-22:
@karawin okay, i will try it at home in evening
@v222008 thanks for your reply. I've connect the node with micro-usb cable and flash with following settings:

[image]

lyovav, 2016-11-22:
And how looks  the module nodeMCU 1.0  that you flashing?

xCite1986, 2016-11-22:
okay, looks good with disconnected VS1053-Board

 ```
=> extraram state: 0 0xff ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea9a8, len: 8392
bss   : 0x3ffea9a8 ~ 0x3fff1850, len: 28328
heap  : 0x3fff1850 ~ 0x40000000, len: 59312
Heap size: 31528
t0 task: 3fff8688
t1 task: 3fff8820
t4 task: 3fff8c48
t3 task: 3fff9268
t2 task: 3fff9f90
mode : sta(5c:cf:7f:89:2b:92)
add if0
I2S Speed: 0
SCI_Mode (0x4800) = 0xFFFF
SCI_Status (0x48) = 0xFFFF
VS Version (VS1053 is 4) = 15
SCI_ClockF = 0xFFFF
UART READY TO READ
first use
 Station Ip: 192.168.1.254
DHCP: 0x1
Device: Ip: 192.168.1.254

I: 0 status: 0
In I: 0 status: 0
In I: 1 status: 0
In I: 2 status: 0
In I: 3 status: 0
In I: 4 status: 0
In I: 5 status: 0
In I: 6 status: 0
In I: 7 status: 0
In I: 8 status: 0
In I: 9 status: 0
In I: 10 status: 0
In I: 11 status: 0
In I: 12 status: 0
In I: 13 status: 0
In I: 14 status: 0

Config not found

The default AP is  WifiWebRadio. Connect your wifi to it.
Then connect a webbrowser to 192.168.4.1 and go to Setting
Erase the database and set ssid, password and ip's field
del if0
usl
sul 0 0
mode : softAP(5e:cf:7f:89:2b:92)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
autostart: playing:255, currentstation:65535
I2S Speed: 0
```

xCite1986, 2016-11-22:
so, if i connect DREQ to D2 it stops with ```add if0 ```...

karawin, 2016-11-22:
So check your dreq wire
(and all other)

karawin, 2016-11-22:
Shorted to the ground....? or 5v or gnd not connected?

xCite1986, 2016-11-22:
yes, everything connected fine. 

I check all connections with your soldered pcb from [here ](http://www.instructables.com/file/FB0VT4VITA8WJ66/)... 
, replace every wire, measure 5V and gnd if the VS gets power.. i dont know.. i'm clueless now.. maybe i've got a crappy VS-board... 

but hey - i took a short look in your website and it was the 
most impressive program i've seen on an ESP8266... ;-)

karawin, 2016-11-22:
Measure the voltage between gnd and dreq. It must be 3.3 v
Check if the 5 volt is on the vs1053.
Without power, check for short circuit between all pins, etc

Thanks, yes it is a bit complex, due to the lack of ram, I digged hard to find some tricks ;-)

xCite1986, 2016-11-22:
no, there was 0V between DGND/G and DREQ/D2 :(
5V on the VS was good

karawin, 2016-11-22:
Power on the vs1053 alone and check dreq.
While removed, check the board connectors of the vs for shorted pins
Check check check

closed 2016-12-03 by karawin