karawin/Ka-Radio #11: Play password protected streams and support for https urls

totalizator, 2016-12-11:
I'm trying to play password protected streams. Such url looks like: **user:password@www.radio.org**. It looks like it's not supported right now.

Other thing is - what would be needed to play streams via https?

karawin, 2016-12-11:
> *user:password@www.radio.org <mailto:password@www.radio.org>*. It

No, it is not and will not be, sorry.
This feature need a |WWW-Authenticate: field and base64 cryptage.

Feasable but....
Same for https. Never seen a webradio with https.

|

totalizator, 2016-12-11:
That's OK. 

The question was about my private streams that are intentionally exposed to the internet and using https is the simplest method for securing the url. No worries though. There are plenty of other methods.

Thank you for the clarification.

closed 2016-12-13 by totalizator