karawin/Ka-Radio #12: 'unsigned int index' redeclared as different kind of symbol

arlaor, 2017-01-16:
Hi @karawin 
When trying to load Karadiolcd16x2.ino, to my module ESP-12E, I am receiving the following message, which will not allow me to continue:
```javascript
karadioNokia5110:41: error: 'unsigned int index' redeclared as different kind of symbol

 unsigned index = 0;

          ^

In file included from c:\users\oz\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\stdlib.h:11:0,

                 from C:\Users\oZ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:27,

                 from C:\Users\oZ\AppData\Local\Temp\arduino_build_598767\sketch\karadioNokia5110.ino.cpp:1:

c:\users\oz\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\string.h:55:8: error: previous declaration of 'char* index(const char*, int)'

 char  *_EXFUN(index,(const char *, int));

        ^

D:\ARDUINO\NODEMCU\mp3 sketch\KaRadio\Firmware\Ka-Radio-master\Lcd & remote\karadioNokia5110\karadioNokia5110.ino: In function 'void loop()':

karadioNokia5110:317: error: invalid types 'char [200][char*(const char*, int)]' for array subscript

             line[index] = 0; // end of string

                       ^

karadioNokia5110:318: error: assignment of function 'char* index(const char*, int)'

             index = 0;

                   ^

karadioNokia5110:318: error: cannot convert 'int' to 'char*(const char*, int)' in assignment

karadioNokia5110:35: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

 #define BUFLEN  200

                 ^

D:\ARDUINO\NODEMCU\mp3 sketch\KaRadio\Firmware\Ka-Radio-master\Lcd & remote\karadioNokia5110\karadioNokia5110.ino:322:23: note: in expansion of macro 'BUFLEN'

             if (index>BUFLEN) break; // small memory so small buffer

                       ^

karadioNokia5110:323: error: ISO C++ forbids incrementing a pointer of type 'char* (*)(const char*, int)' [-fpermissive]

             line[index++] = temp;

                       ^

karadioNokia5110:323: error: lvalue required as increment operand

Usando librería EEPROM con versión 1.0 en la carpeta: C:\Users\oZ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\EEPROM 
exit status 1
'unsigned int index' redeclared as different kind of symbol
```
What modification should I make to your sketch in order to use it?

What is the sketch that allows me to handle it only from a web page, without using an LCD screen?

karawin, 2017-01-16:
Karadio is not an arduino IDE software.
Please download http://www.espressif.com/sites/default/files/tools/flash_download_tools_v3.3.4_win.zip
It is the tool needed to flash the esp8266.
In the directory ESP8266-Firmware/bin you will find the files to be flashed.
See
[image] 
To wire your prototype and
[image]
to configure the flasher.

The ino sketch you found is for the external arduino pro mini to add a lcd and IR.
Hope this help.

closed 2017-01-16 by karawin

arlaor, 2017-01-16:
Hi @karawin 
Initially use the tools you have published, but after updating it does not appear by any side some network generated by my module to connect to this or some connection request to connect it to my Wifi.

So, if you do not find an event like the ones mentioned above, try to enter the sketch published on your page karadiolcd16x2.ino

karawin, 2017-01-16:
Try again, many many other have succeded. It was blinking?
On your pc the wifi must be detected id it is blinking.