karawin/Ka-Radio #13: Flashing does not work

Testi89, 2017-01-21:
Hello,
if have downloaded the files from hackaday and flashed like this:
"To upgrade to this release, please flash user1.4096.new.4.bin at 0x1000 ,
user2.4096.new.4.bin at 0X81000 and blank.bin at 0x7e000 & 0x3fe000"
and also boot.1.6.bin to 0x0000

but I do not get a connection. Something is wrong with flashing.

Can you please check and update the flashing guidelines?

Thanks a lot!

karawin, 2017-01-21:
These instructions are for an upgrade.
For a first flash see
https://hackaday.io/project/11570-wifi-webradio-with-esp8266-and-vs1053
or  https://cdn.hackaday.io/images/8829451470433125187.png

closed 2017-01-21 by karawin

Testi89, 2017-01-22:
Hello Jean-Pierre,

merci pour le support.

I flashed again and now I get this result:
=> extraram state: 1 0x30 0123456789ABCDEF
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea9f8, len: 8472
bss   : 0x3ffea9f8 ~ 0x3fff18a8, len: 28336
heap  : 0x3fff18a8 ~ 0x40000000, len: 59224
Heap size: 44408
t0 task: 3fff5438
t1 task: 3fff55d0
t4 task: 3fff59f8
t3 task: 3fff6018
t2 task: 3fff6d68
mode : sta(a0:20:a6:0b:e7:05)
add if0

I cannot see any new WiFi spot to log into.

karawin, 2017-01-22:
Nothing else in log?
try a sys.erase 
and reboot

Testi89, 2017-01-22:
I have downloaded the 6 files from the hackaday page and erased the chip:

$ esptool.py erase_flash
esptool.py v1.3-dev
Connecting...
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 10.3 seconds

Then I run flashing:

$ esptool.py write_flash 0x000 boot_v1.6.bin 0x01000 user1.4096.new.4.bin 0x81000 user2.4096.new.4.bin 0x7e000 blank.bin 0x3fe000 blank.bin 0xfc000 esp_init_data_default.bin 0x3fc000 esp_init_data_default.bin 
esptool.py v1.3-dev
Connecting...
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.4 kbit/s)...
Writing 380928 @ 0x1000... 380928 (100 %)
Wrote 380928 bytes at 0x1000 in 33.0 seconds (92.3 kbit/s)...
Writing 380928 @ 0x81000... 380928 (100 %)
Wrote 380928 bytes at 0x81000 in 33.0 seconds (92.3 kbit/s)...
Writing 4096 @ 0x7e000... 4096 (100 %)
Wrote 4096 bytes at 0x7e000 in 0.4 seconds (89.6 kbit/s)...
Writing 4096 @ 0x3fe000... 4096 (100 %)
Wrote 4096 bytes at 0x3fe000 in 0.4 seconds (89.7 kbit/s)...
Writing 4096 @ 0xfc000... 4096 (100 %)
Wrote 4096 bytes at 0xfc000 in 0.4 seconds (89.7 kbit/s)...
Writing 4096 @ 0x3fc000... 4096 (100 %)
Wrote 4096 bytes at 0x3fc000 in 0.4 seconds (89.7 kbit/s)...
Leaving...

Output is:
=> extraram state: 1 0x30 0123456789ABCDEF
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea9f8, len: 8472
bss   : 0x3ffea9f8 ~ 0x3fff18a8, len: 28336
heap  : 0x3fff18a8 ~ 0x40000000, len: 59224
Heap size: 44408
t0 task: 3fff5438
t1 task: 3fff55d0
t4 task: 3fff59f8
t3 task: 3fff6018
t2 task: 3fff6d68
mode : sta(a0:20:a6:0b:e7:05)
add if0

Maybe the downladed files are not fully correct?

Testi89, 2017-01-22:
After serveral resets I got this output and can connect to the 192.168.1.4:
=> extraram state: 1 0x30 0123456789ABCDEF
data  : 0x3ffe8000 ~ 0x3ffe88d4, len: 2260
rodata: 0x3ffe88e0 ~ 0x3ffea9f8, len: 8472
bss   : 0x3ffea9f8 ~ 0x3fff18a8, len: 28336
heap  : 0x3fff18a8 ~ 0x40000000, len: 59224
Heap size: 44408
t0 task: 3fff5438
t1 task: 3fff55d0
t4 task: 3fff59f8
t3 task: 3fff6018
t2 task: 3fff6d68
mode : sta(a0:20:a6:0b:e7:05)
add if0
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0x48
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0x6000
I2S Speed: 0
UART READY TO READ
first use
 Station Ip: 192.168.1.254
DHCP: 0x1
Device: Ip: 192.168.1.254

I: 0 status: 0
In I: 0 status: 0
In I: 1 status: 0
In I: 2 status: 0
In I: 3 status: 0
In I: 4 status: 0
In I: 5 status: 0
In I: 6 status: 0
In I: 7 status: 0
In I: 8 status: 0
In I: 9 status: 0
In I: 10 status: 0
In I: 11 status: 0
In I: 12 status: 0
In I: 13 status: 0
In I: 14 status: 0

Config not found

The default AP is  WifiWebRadio. Connect your wifi to it.
Then connect a webbrowser to 192.168.4.1 and go to Setting
Erase the database and set ssid, password and ip's field
del if0
usl
mode : softAP(a2:20:a6:0b:e7:05)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
autostart: playing:255, currentstation:65535
I2S Speed: 0
add 1
aid 1
station: 1c:91:48:9e:71:e0 join, AID = 1

Thanks a lot.

karawin, 2017-01-22:
It seems that it is not the last release.
Now do an OTA upgrade in the setting panel.
Great.

Testi89, 2017-01-22:
Now I can hear music. Great!
Thanks a lot for this great software and keep up the superior work!

zigithu, 2017-06-04:
I have had same troubles with flashing esp12e.
Karadio has started to work when i have changed SPI mode from QIO to DIO before flashing (despite of window "detected info" says that my ESP has flash with QUAD) 
[image]

karawin, 2017-06-04:
Is it a ai_thinker or a clone?

zigithu, 2017-06-06:
i have bought it here: https://ru.aliexpress.com/item/ESP8266-CH340G-CH340-G-NodeMcu-V3-Lua-Wireless-WIFI-Module-Connector-Development-Board-Based-ESP-12E/32800966224.html?

doctormord, 2017-07-14:
I'm having the same issues. The latest 1.3.4 code wont boot on a Wemos D1 Mini or NodeMCU. I tried flashing under Windows and Ubuntu. After Reset i onle get garbage in the console. Same on both boards. Anything i missed? Flash is set up as 32Mbit QIO, but even DIO wont do. (With and without VS1053 connected)

pmr446dx, 2017-09-11:
The only ESP Download tool version currently available for download on Espressif website is version 3.4. This looks quite different from 2.3 presented in the instructions. Was anyone able to flash using v 3.4?
Thanks

karawin, 2017-09-11:
3.3.4 3.4.4 any tools are good to flash.

pmr446dx, 2017-10-21:
Thank you Jean-Pierre.
I was able to flash using the latest versionof the ESP Download tool
I flashed on a "Wemos D1 mini Pro" with 16 MB  of RAM (schematics [here](https://wiki.wemos.cc/products:d1:d1_mini_pro)).
The software boots correctly and initialises as an access point, I am able to browse to 192.168.4.1 and set up the SSID and password of my AP, but the radio does not save the settings after restart.
Do you know if there is any chance to make Ka-Radio work on this 16MB board?

karawin, 2017-10-21:
Oups, i forgot to update the repo.
Try now with the new bin.
Or download they on 
http://karadio.karawin.fr/user1.4096.new.4.bin
http://karadio.karawin.fr/user2.4096.new.4.bin

I think it will works now

pmr446dx, 2017-10-22:
You are right, it worked perfectly with the new bin files.
This is very strange to me, because this happens since the first time I tried, 11 September.
I cannot believe that no one else noticed it on the forums or facebook page. :) 
Maybe they had the software already deployed and only used OTA upgrade.
Thank you! I also got esp32 karadio working on 1st try, version 0.3. I will upload some pictures of my projects as soon as they are finished :)
Best Regards.
Florin

karawin, 2017-10-22:
You are not the first one but the second one.
In fact, i made the correction in beginning of september and forgot to update github.
Happy it works
Bye

ps try the update to 0.4 on KaRadio32. Very stable.

arlaor, 2018-01-17:
Hi @karawin 
Excuse me
I saw that they talked about a wemos
To use the project in a wemos mini pro, should some parameters of the flash download tool be changed? when using it with the parameters of the image it generates an error with strange symbols and fast flashing of the led.
I used the bin files 1.6.3

[image]

arlaor, 2018-01-18:
[image]

karawin, 2018-01-18:
arlaor, send the log of the uart on boot please.

arlaor, 2018-01-19:
Hi @karawin 
atach the image arduino info
It has not allowed me to copy the text generated in the IDE
[image]

other observation: In DETECTED INFO, QUAD;128MB
this is normal?
[image]

karawin, 2018-01-19:
The detected info is normal.
Correct.
Just for esp_init_data_default.bin, remove the 0xFC000 and replace to 0x3FC000
Check the files length.
Cannot tell you other things.

arlaor, 2018-01-20:
Hi @karawin 

I have seen that the new version of the download tool V3.6.2.2 needs a line of address and is left out 0x3FC000.
This work
Thank you.