karawin/Ka-Radio #15: Connections unclear

Testi89, 2017-01-31:
Hello,
can you please check the documentation for the connections? I think in the PDF from Piotr Sperka there are other connections that with you readme.md file. Also the RAM is not mentioned in the readme.
Thanks a lot.

karawin, 2017-01-31:
Yes, i will try.
It is a lot of work to do a document and the software and answer to all 
questions.
regards

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

Testi89, 2017-02-13:
Can you please check if the 28800 baud for LCD is correct (see code and documents)? From ESP8266 I get readable output at 115200 baud.
Merci!

karawin, 2017-02-13:
On the uart interface type
sys.uart("28800")
to change the speed of the interface.

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

arlaor, 2017-03-06:
Hi Karawin. One Question. ¿Using a wemos mini PRO module, 16MB flash improves reception of stations with 192kbps or 320kbps? Or is it the same as using an ESP-12E module ?, I have problems cutting the signal with stations of these bitrates

karawin, 2017-03-07:
Hi

The same.
Some 320kbps doesn't work, some works. Cannot find why.
192 works in all case with esp12E

NonaSuomy, 2017-04-09:
I did some fritzing breadboard/schematics
[image]
[image]
[image]
[image]

arlaor, 2017-04-20:
Hello @karawin , I found [this](https://www.aliexpress.com/item/NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-flash-USB-serial-CH340G/32791599265.html?spm=2114.13010208.99999999.264.RXGuCv) module, which says to have 32MB of memory. When using this module, can this improve the operation of the project?

karawin, 2017-04-20:
32Mbits equal 4Mbytes. Karadio is generated for that. 
Using a 16MB may be a problem for ota update. I use only 32MB esp12e, bare one or in a nodeMCU pcb.

arlaor, 2017-04-20:
I understand, then it's just a game of numerical values.
There is nothing left but to expect that there is enough support for the ESP32

NonaSuomy, 2017-05-04:
Does freeRTOS just have too much overhead? The Arduino project seems to stream smoothly with no ext ram [ESP-Radio](https://github.com/Edzelf/Esp-radio) just a ring buffer up to 320kbps@180mhz / 256kbps@80mhz

karawin, 2017-05-04:
Karadio does not need extra ram to run.
It just serves to improve the reception problems of the html stream.
Freertos is much better than a simple loop.

closed 2017-05-24 by karawin