karawin/Ka-Radio #16: Fail list Station

arlaor, 2017-02-10:
Hi @karawin 
I am trying to load this [List](https://gist.github.com/arlaor/cf9a045099a008104cb3d368cddaedc6) of stations, but after having several restarts of the module, the list appears in the web page of the project.
When playing the stations, 50% of them work and the other 50% do not.
When I press "Refresh locale list" on the STATIONS website, 50% randomly disappears, which are the ones that can not be played when they appear in the drop-down list.
I'm using version 1.1.7
The same thing happened in previous versions

karawin, 2017-02-10:
Yes, the correction was not well done.
Update again please and let me know.
Thanks

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

arlaor, 2017-02-10:
I just updated now and the error persists

karawin, 2017-02-10:
May be a cache pb. V1.1.7+ must be on top of the web page.

arlaor, 2017-02-10:
Now I see 1.1.7+
Now only I lose the last station in the list and the volume preset in some stations, is formatted. This volume has also been saved in the txt file. All preset volumes are converted to zero

karawin, 2017-02-10:
test

Not correct?
For the last station, yes, it is better to keep all empty stations in 
the list to erase existing ones. They are loaded Two by two, and i 
assume that the list is as when saved.

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

karawin, 2017-02-10:
[http://karadio.karawin.fr/karadio/images/screenshot_20170210_180113.png](http://karadio.karawin.fr//karadio/images/screenshot_20170210_180113.png)

arlaor, 2017-02-10:
Is strange, in my list disappears the volume saved.
No radio station is deleted anymore

karawin, 2017-02-10:
Check your txt file?

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

arlaor, 2017-02-10:
I have tried reloading the WebStations.txt file generated by the web page from the "Save Stations from WebRadio" button, which saves all the volume parameters that are observed the first time I upload my txt file, but when I do " Refresh locale list (i)" disappear

karawin, 2017-02-10:
Congrat, you have found a bug. I missed to add the volume offset in the refresh list.
I will correct and let you know.

karawin, 2017-02-10:
Ok now.
Don't forget to refresh the page in the browser.
Thanks.

arlaor, 2017-02-10:
I get it.
I'm trying to play these stations but I can not play them
http://streams.deltaradio.de/delta-alternative/mp3-128/
http://streams.deltaradio.de/delta-foehnfrisur/mp3-128/
http://streams.deltaradio.de/delta-indie/mp3-128/
http://streams.deltaradio.de/delta-unplugged/mp3-128/
http://streams.deltaradio.de/delta-poprock/mp3-128/
http://streams.deltaradio.de/delta-sommer/mp3-128/

These others are from the same radio station and work normally:
http://streams.deltaradio.de/delta-live/mp3-128/
http://streams.deltaradio.de/delta-grunge/mp3-128/
http://streams.deltaradio.de/100/mp3-128/
http://streams.deltaradio.de/101/mp3-128/

arlaor, 2017-02-10:
The list of radio stations now works correctly. The volume saved in the list is not cleared

karawin, 2017-02-10:
"I'm trying to play these stations but I can not play them"
Yes the redirection is very long. I increased the buffer. Now they works.
Thanks again.

arlaor, 2017-02-10:
Thanks to you for this excellent job

karawin, 2017-02-10:
Can i link your playlist on the Setting panel?

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

arlaor, 2017-02-10:
Of course, but there is a problem, the stations that belong to:
Radiotunes.com, Di.fm and rockradio.com are payable and the link expires in approximately one month. The other stations have no expiration problem.

karawin, 2017-02-10:
Ok, I will make a note to explain it.
I close the issue.

closed 2017-02-10 by karawin

arlaor, 2017-02-10:
If you want I can filter the payment stations and leave those that will have normal functionality

karawin, 2017-02-10:
great, thanks

arlaor, 2017-02-10:
Here is the [link](https://gist.github.com/arlaor/cf9a045099a008104cb3d368cddaedc6)

karawin, 2017-02-10:
Updated, thanks