karawin/Ka-Radio #18: Autostart does not work (1.1.7+)

Testi89, 2017-02-17:
Hello @karawin ,
When I startup my module, the autostart is not working, means that the radio station is not played.
I have to manually press "start" from the webpage. Volume is also not saved.

dhcp client start...
ip:192.168.1.30,mask:255.255.255.0,gw:192.168.1.1
autostart: playing:255, currentstation:0
I2S Speed: 0

After pressing "play":
##CLI.STOPPED# from playStationInt
##CLI.NAMESET#: 0 Alsterradio
##CLI.URLSET#: live96.alsterradio.de
##CLI.PATHSET#: /
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 144
Header: Moved
##CLI.STOPPED# from C_HEADER
##CLI.URLSET#: stream.alsterradio.de
##CLI.PATHSET#: /alsterradio/mp3-128/
##CLI.PORTSET#: 80
Header: Moved
##CLI.STOPPED# from C_HEADER
##CLI.URLSET#: 138.201.251.245
##CLI.PATHSET#: /alsterradio_128
##CLI.PORTSET#: 80
##CLI.ICY0#: alsterradio rock 'n pop
##CLI.ICY3#: http:\/\/www.alsterradio.de
##CLI.ICY4#: Rock Pop
##CLI.ICY5#: 128
##CLI.ICY6#: Playlist
##CLI.META#: TOTO - HOLD THE LINE
##CLI.PLAYING#

karawin, 2017-02-17:
autostart: playing:255
Please uncheck and check again autoplay check box.
The first time it needs it.

Testi89, 2017-02-17:
Yes, now it is working. Is this needed after every firmware upgrade?
The checkbox on the webpage was correctly marked "on".

karawin, 2017-02-17:
No need

17 févr. 2017 a écrit :
>Yes, now it is working. Is this needed after every firmware upgrade?
>The checkbox on the webpage was correctly marked "on".

-- 
Envoyé de mon téléphone Android avec <a href="https://play.google.com/store/apps/details?id=com.onegravity.k10.pro2"><b>K-@ Mail</b></a>. Excusez la brièveté.

closed 2017-02-17 by Testi89