karawin/Ka-Radio #19: Button control

tomasf71, 2017-02-17:
I tried to use button control via ADC input. I follow your schematic (resistive divider 1k, 1k, ..., 16k)
but some buttons doesn't work. Next station, Start button.  What is ADC thresholds for individual buttons. I am going to try measure voltage on ADC input.

Testi89, 2017-02-17:
Which voltage do you use?

karawin, 2017-02-17:
The feature:
*What is the range of the NodeMCU ADC input?*
Measured range: 0 - Vdd. (About 3.1 V in this case)
Number of bits: 10 bit (Codes: 0-1024)

Note that the NodeMCU board has a voltage divider on the ADC input. The 
ESP8266 module it self has 0 - 1.0 V input range.

Readings are done with the command: print(adc.read(0))
0.5 V => 167
1.0 V => 330
1.5 V => 498
2.0 V => 666
2.5 V => 828
3.0 V => 995
3.1 V => 1024 (saturated)

The karadio software check if the reference is 3.1 or 1.0 v automatically.
     if (system_adc_read() < 10) adcdiv = 0; // no panel adc grounded
     else
     // read adc to see if it is a nodemcu with adc dividor
         if (system_adc_read() < 400) adcdiv = 3;
             else adcdiv = 1;

The code for buttons:
void switchCommand() {
     int adc;
     char Vol[22];
     if (adcdiv == 0) return; // no panel

         adc = system_adc_read();
         adc *= adcdiv;
//        if (adc < 930)
//            printf("adc: %d  div: %d\n",adc,adcdiv);
//        if (adc >940) vTaskDelay(10);
         if (inside&&(adc > 930)) inside = false;

         if ((adc >400) && (adc < 580)) // volume +
         {
             setVolumePlus();
         }
         else if ((adc >730) && (adc < 830)) // volume -
         {
             setVolumeMinus();
         }
         if (!inside)
         {
             if (adc < 220) // stop
             {
                 inside = true;
                 clientDisconnect("Adc Stop");
             }
             else if ((adc >278) && (adc < 380)) //start
             {
                 inside = true;
                 sprintf(Vol,"%d",currentStation);
                 playStation    (Vol);
             }
             else if ((adc >830) && (adc < 920)) // station+
             {
                 inside = true;
                 wsStationNext();
             }
             else if ((adc >590) && (adc < 710)) // station-
             {
                 inside = true;
                 wsStationPrev();
             }
         }
}

May be the 16k needs to be adjusted to have 1.0 v when no button are 
pressed.

tomasf71, 2017-02-17:
Thank you. I use 3.3VT. I am going to try adjust R (16k) to have 1V on ADC input when no button pressed.

zajaklis, 2017-02-18:
i had same problem before i add additional 1k to 16k resistor. otherwise there was small offset for some buttons. now all ok.

tomasf71, 2017-02-18:
I did some simple test. To read voltage and RAW data on A0. Here is the result: 
[image]

So, if no button pressed, there is 0,87V. It is what I supposed:  (3.3V / (16k+6k))*6k = 0,9 V

tomasf71, 2017-02-18:
So, it looks to add 2k resistor in series with 16k should do the job.

karawin, 2017-02-18:
> snimek obrazovky 2017-02-18 v 16 05 26

No
(3.3V / (16k+7k))*7k = 1.00434783V

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

karawin, 2017-02-18:
---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

tomasf71, 2017-02-19:
You are right :-)

closed 2017-02-19 by karawin