karawin/Ka-Radio #20: Nextion hmi

AST-esp-rasp, 2017-02-28:
Hello,
You have done a great job!!!
Controlling Kia radio with nextion hmi seems to work (waiting for my display used simulation softwre from nextion). Noe I'm wondering if some additional printf commands in webserver and client would allowing to get correct feedback for the display. Actually i could see there ##CLI.xxx# strings printed.
Would it harm the code when im adding an additional printf with my nextion instruction string?
And of course it only works when i figure out how to compile it?

karawin, 2017-03-15:
see http://karadio.karawin.fr/Interface.txt for the complete list of commands and prints

closed 2017-03-15 by karawin

karawin, 2017-03-17:
Just add a 3$ µc with 2 serial interfaces and write your own software. It is very easy and much more easy to debug and update.

AST-esp-rasp, 2017-03-22:
That's true! It will save a lot of effort! 
But it won't be the most sophisticated solution.