karawin/Ka-Radio #21: Add PWM_HACK for testing from ESP8266_MP3_DECODER

NonaSuomy, 2017-03-13:
Hi can you please add this option, for people that just want to test it out before they commit to buying an i2s dac.

Thank you.

karawin, 2017-03-15:
No i don't implement it. Bad quality in mono mp3 and I don't want to waste the precious ram.
To test the software, just avoid the vs1053. 
The software run without it, but no sound obviously.
Thanks

closed 2017-03-15 by karawin

NonaSuomy, 2017-03-17:
You won't waste any precious ram what so ever to have that option available, you would just do something like this;

At the top of the source file. Where you have some of your variable options, surround it with a preprocessor directive:

```
//#define PWMHACK
// ...More code...
#ifdef PWMHACK
  // "Debug PWMHACK enabled..."
  // More code etc...
#endif
```

at compile time none of this code will be included unless you uncomment //#define PWMHACK => #define PWMHACK . I can't test through the current interface because I want to stream custom audio samples to the device which will be text to speech voice prompts from a home automation system.

karawin, 2017-03-17:
I tested a text to speech with http://www.voicerss.org/ with success.
Sorry but the software investment is to heavy for an under 10$ hardware replacement.
But the code is freely available go on!
I see that you have some knowing in programming ;-)
I am just a software engineer since 1980.

NonaSuomy, 2017-03-19:
What did you do for the test of voicerss.org?

I thought you said someplace that you based this on esp8266_mp3_encoder so I didn't know that you changed it much from their implementation, thought it would just be a pull of the original base to get it working with yours. I already bought the hardware but it will take over a month to get here before I can do any proper testing with OpenHab.

My programming experience is of a hacker level, smashing code together and hoping for the best, so I am no place near your level, Unfortunately, something that would take you seconds would take me years.

[PWM_HACK](https://github.com/espressif/ESP8266_MP3_DECODER/blob/master/mp3/user/user_main.c#L65)
Is the source no longer alike from it's base?

karawin, 2017-03-19:
I tested with http://192.168.1.253/?instant="http://api.voicerss.org/?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello?the-sun-is-yellow?and-i-like-it"

KaRadio has nothing in common with PWM_HACK.
So the integration is a very big work.

NonaSuomy, 2017-04-05:
Thanks for the test link, worked.