karawin/Ka-Radio #24: Connected SRAM IC doesn't seem to respond.

NonaSuomy, 2017-04-05:
When I type sys.ram, should I see some info, if there is no ram chip available or it only shows something if there is external ram hooked up.

I Connected external ram 23LC1024.
[23LC1024](http://www.microchip.com/wwwproducts/en/23LC1024)
[23LCV1024](http://www.microchip.com/wwwproducts/en/23LCV1024)
[PDF 23LC1024](http://ww1.microchip.com/downloads/en/DeviceDoc/20005142C.pdf)
[PDF 23LCV1024](http://ww1.microchip.com/downloads/en/DeviceDoc/25156A.pdf)

23LC1024 has Vhold <== using this one.
23LCV1024 has VBAT

*Note: If your chip has a /Vhold in place of /VBAT, the pin 7 must be wired to VCC (pin8)* instead of pin 7 to GND for the V series..

1:CS = GPIO16 
2:MISO = GPIO12 
3:NC 
4:GND 
5:MOSI = GPIO13 
6:SCK = GPIO14 
7:VCC 3.3v 
8:VCC 3.3v

Double checked wiring still no love.
[image]

WebRadio with the VS1053 works so far only at lower bitrates so far 24k streams from stream6.jungletrain.net:8024 is the first one I got working so far with no stuttering, 64k stutters stream6.jungletrain.net:8064, 128k is worse stream6.jungletrain.net:8128 hopefully the ram helps?

karawin, 2017-04-05:
> img_20170404_223418

the command sys.ram doesn't exist.
To see if the external ram is detected, see the log after a reset:
=> extraram state: 0 0xff ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ  show a non detected ram
=> extraram state: 1 0x30 1234567890  show a detected ram

Be sure to wire a0 (adc0) to ground.

=> extraram state: 0 0xff ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
data  : 0x3ffe8000 ~ 0x3ffe8902, len: 2306
rodata: 0x3ffe8910 ~ 0x3ffead7c, len: 9324
bss   : 0x3ffead80 ~ 0x3fff1bc0, len: 28224
heap  : 0x3fff1bc0 ~ 0x40000000, len: 58432
Heap size: 30648
t0 task: 3fff89f8
t1 task: 3fff8b90
t4 task: 3fff90e8
t3 task: 3fff9730
t2 task: 3fffa480
mode : sta(5c:cf:7f:1c:75:ab)
add if0
UART READY TO READ
AP1: karawin2, AP2:
  Station Ip: 192.168.1.253
scandone
Trying karawin2,  I: 0 status: 1
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0x4C
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0xB800
I2S Speed: 0
plugin size 18ce 2 c67
plugin start: 7 1 8050 6
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
pm open phy_2,type:2 0 0
cnt

connected with karawin2, channel 11
dhcp client start...
ip:192.168.1.253,mask:255.255.255.0,gw:192.168.1.70
autostart: playing:1, currentstation:23
I2S Speed: 0

NonaSuomy, 2017-04-05:
Interesting, I could have sworn I saw that in this [interface.txt](http://karadio.karawin.fr/Interface.txt) guess I was just seeing what I wanted to :smile_cat: 

Seems it detects it fine, I guess I just expected too much, that it was going to solve the constant stutter :disappointed: 

Any way to figure out what is wrong with any streams above 24kbps.

Using an Adafruit Huzzah (ESP8266).

50Mbps connection, any way to do a speed test on the module?

 ets Jan  8 2013,rst cause:1, boot mode:(3,5)

load 0x40100000, len 1856, room 16 
tail 0
chksum 0x63
load 0x3ffe8000, len 776, room 8 
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8 
tail 0
chksum 0x79
csum 0x79

2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user2 @ 81000

~
 lûÁn

    llì
       àl`nÄÞRÿ
=> extraram state: 1 0x30 0123456789ABCDEF
data  : 0x3ffe8000 ~ 0x3ffe8902, len: 2306
rodata: 0x3ffe8910 ~ 0x3ffead7c, len: 9324
bss   : 0x3ffead80 ~ 0x3fff1bc0, len: 28224
heap  : 0x3fff1bc0 ~ 0x40000000, len: 58432
Heap size: 43616
t0 task: 3fff5750
t1 task: 3fff58e8
t4 task: 3fff5e40
t3 task: 3fff6488
t2 task: 3fff71d8
mode : sta(18:fe:34:a5:c5:b7)
add if0
UART READY TO READ
AP1: All Your Base, AP2: 
 Station Ip: 10.13.38.65
scandone
Trying All Your Base,  I: 0 status: 1
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0x4C
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0xB800
I2S Speed: 0
plugin size 18ce 2 c67 
plugin start: 7 1 8050 6
scandone
state: 0 -> 2 (b0)
Trying All Your Base,  I: 1 status: 1
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
pm open phy_2,type:2 0 0
cnt 

connected with All Your Base, channel 9
dhcp client start...
ip:10.13.37.65,mask:255.255.255.0,gw:10.13.37.1
autostart: playing:0, currentstation:113
I2S Speed: 0

NonaSuomy, 2017-04-16:
Did some Fritzing breadboard/schematics for the two different SRAM connections.

I used a Geeetech VS1053B breakout from ebay which was cheaper than the sparkfun breakout.

**23LC1024 SRAM**
[image]
[image]

**23LCV1024 SRAM**
[image]
[image]

closed 2017-05-30 by karawin