karawin/Ka-Radio #25: Python script to send urls from streamtheworld.com to player.

NonaSuomy, 2017-04-06:
<script src="https://gist.github.com/NonaSuomy/e32ff94ed728c8d31947477787d59d89.js"></script>

desadsea, 2017-04-21:
Hi, sorry but I did not understand how to use the script, could you make a practical example? Thanks Ezio

NonaSuomy, 2017-04-22:
@desadsea it's very simple at the top of the script is the usage example, basically find a radio station on iheartradio [iheartradio station list](https://en.wikipedia.org/wiki/List_of_radio_stations_owned_by_iHeartMedia)
```Usage: python2 streamtheworldka.py CIMXFM 10.13.37.65
                                   ^Radio callsign  ^IP address of Ka-Radio Device
```
make sure you have python2 and curl installed on the system you are sending the station from, download the script to a file on that device, then type the command, I named the script streamtheworldka.py, so you type on the command line python2 streamtheworldka.py (The radio call sign + FM on the end) (IP Address of your Ka-Radio Player)

First, test with the call sign above/below make sure it is all in CAPS as I know that one works (may not work outside of Canada), the hard part would be finding how they named the call sign for the station you want, and some may be location locked so be aware, you also have to add that FM/AM on the end of the call sign.

The script then tells the website to give us the ShoutCast stream URL for audio stream compatibility purposes with _SC.

```
python2 streamtheworldka.py CIMXFM 10.13.37.65
```

The script is grabbing the stream URL that is randomly encoded a string of characters for that station you want then sending the URL to the ESP8266.

The script request a URL like this:
http://playerservices.streamtheworld.com/api/livestream?version=1.5&mount=CIMXFM&lang=en

Then it sends back an XML which we extract the values from:
```
<live_stream_config xmlns="http://provisioning.streamtheworld.com/player/livestream-1.5" version="1.5">
  <script/>
    <mountpoints>
      <mountpoint>
        <status>
          <status-code>200</status-code>
          <status-message>OK</status-message>
        </status>
        <transports>
          <transport>http</transport>
        </transports>
        <metadata>
          <shoutcast-v1 enabled="true" mountSuffix="_SC"/>
          <shoutcast-v2 enabled="false" mountSuffix="_SC"/>
          <sse-sideband enabled="true" streamSuffix="_SC" metadataSuffix="_SBM"/>
        </metadata>
        <servers>
          <server sid="18853">
          <ip>18853.live.streamtheworld.com</ip>
          <ports>
            <port type="http">80</port>
            <port type="http">3690</port>
          </ports>
        </server>
        <server sid="17023">
          <ip>17023.live.streamtheworld.com</ip>
          <ports>
            <port type="http">80</port>
            <port type="http">3690</port>
          </ports>
        </server>
        <server sid="13733">
          <ip>13733.live.streamtheworld.com</ip>
          <ports>
             <port type="http">80</port>
             <port type="http">3690</port>
          </ports>
        </server>
        <server sid="14213">
          <ip>14213.live.streamtheworld.com</ip>
          <ports>
            <port type="http">80</port>
            <port type="http">3690</port>
          </ports>
        </server>
      </servers>
      <metrics>
        <tag name="uuid"/>
      </metrics>
      <mount>CIMXFM</mount>
      <format>FLV</format>
      <bitrate>48000</bitrate>
      <media-format container="flv" cuepoints="stwcue">
        <audio index="0" samplerate="22050" codec="mp3" bitrate="48000" channels="2"/>
      </media-format>
      <authentication>0</authentication>
      <timeout>0</timeout>
      <send-page-url>0</send-page-url>
    </mountpoint>
  </mountpoints>
</live_stream_config>
```

The script extracts,

Random generated IP: 18853.live.streamtheworld.com
Port: 80, 3690, etc
MountPoint(callsign): CIMXFM

I noticed that there was ShoutCast compatibility in the XML which it shows to add "_CS" onto the mount point: CIMXFM_CS

Then the script sends this data gathered from the XML to Ka-Radio device:

curl http://10.13.37.65/?instant="18853.live.streamtheworld.com:3690/CIMXFM_SC"

Then Ka-Radio should start playing the stream.

If you attach a serial cable to Ka-Radio you should be able to debug what URL is getting passed to it, to see if the station is getting sent properly.

Hope it helps!

NonaSuomy, 2017-04-23:
@desadsea I have to thank you, I clicked on your name and you had a repository for esp-radio with Arduino, been hacking away at it and got it loaded, radio stations stream flawlessly where I couldn't get Ka-Radio to stream anything over 24Kbps. It does 320kbps@160Mhz 256kbps@80Mhz, Also, it uses no EXT RAM just a Ring Buffer. [Esp-radio](https://github.com/Edzelf/Esp-radio) Guess I will be switching over to it for now until I find an issue with it that I can't solve like stuttering with Ka-Radio. Will have to try this script with it as well :+1:

desadsea, 2017-04-23:
@NonaSuomy Thank you for having responded, and helped me to figure out how Python script works,
As soon as I have the time I want to try, with regard to the esp with arduino, it's not my project I put it on githbub and how would you like to try it, thank you again for your kindness Ezio.

NonaSuomy, 2017-05-04:
Did an [OpenHAB mod for ESP-Radio](https://github.com/Edzelf/Esp-radio/issues/77)

NonaSuomy, 2017-05-04:
And integrated the xml parser into ESP-Radio for [IHeartRadio](https://github.com/NonaSuomy/Esp-radio/tree/NonaSuomy-XML-Parser)

karawin, 2017-05-04:
What is the interest of streamtheworld?

NonaSuomy, 2017-05-06:
Just to note the parser I did was included into the main branch of ESP-Radio now for anyone interested presets look like ihr/CIMXFM.

StreamTheWorld consumes local radio station content and then they rebranded the site as iHeartRadio.

Some people enjoy local radio content stations for stuff relative to them or feels of nostalgia.

[wiki iHeartRadio](https://en.wikipedia.org/wiki/IHeartRadio)