karawin/Ka-Radio #26: How to connect I2C display and hardware buttons?

W1nterSilence, 2017-04-09:
Hi! First of all I would like to say thanks a lot for this project, - it is exactly what I was looking for to build up a WiFi radio.

Sorry this issue thing is not supposed to ask this kind of questions, I am not so familiar with the GitHub communication principles and approaches, but I saw in comments on [instructables](http://www.instructables.com/id/Wifi-Webradio-With-ESP8266-and-VS1053/?ALLSTEPS) and [Hackaday](https://hackaday.io/project/11570-wifi-webradio-with-esp8266-and-vs1053) that there is some way to add LCD and buttons. Do you have instructions to cover how it could be implemented?

And one more - I saw the instructions to upload your firmware via the ESPRESSO. I guess that the same could be done with the source code and Arduino SDK, but can't find out how it can be done exactly. Do you have some docs to cover this part as well?

Again, - thanks again for this great project.

karawin, 2017-04-09:
Hi,

I made 4 projects to add a lcd  with or without Remote IR control.
See them at 
karadio.karawin.fr/karadioU8glib.zip  for black and white lcd or oled
karadio.karawin.fr/karadiolcd16x2.zip for a 16x2 display
karadio.karawin.fr/karadioUCglib.zip For a color lcd
karadio.karawin.fr/karadioUCSTM32.zip the new one with color lcd, IR and a rotary encoder. 
 Very easy. Turn knob= volume. One click+turn=station choice, double click= start/stop.
 See https://youtu.be/_kqOC5ySlSM

For each: Description in zip ucglibConf.h file, source for arduino.

What is espresso?
KaraDio itself is not an arduino project.
Thanks

W1nterSilence, 2017-04-09:
Sorry, I've mistyped - I meant [Espressif Flash Download Tool](http://bbs.espressif.com/viewtopic.php?t=641). Since it' s GUI version is available only for Windows, I thought that I could use Arduino to upload firmware)

Am I getting it right that if I want to use LCD and keyboard, - I should add arduino to handle it? Well, I  believe that it is quite obvious question, but nevertheless - is there any chance to use LCD directly to ESP8266?

karawin, 2017-04-09:
It exists various way to flash the esp including arduino i suppose.
Anyway, you have to flash it only one time because all update will be with ota directly.

I don't want to use a lcd directly. It is too restrictive. I offer a clean interface on uart for an external cpu.
Much more flexible. Karadio is big and difficult to install and generate. An exteraznl cpu under arduino is simple and flexible.

closed 2017-04-18 by karawin