karawin/Ka-Radio #28: After enter SSID - ESP dead?

olemde, 2017-04-20:
Hi.
i have flashed the binary and had a connection to the "WifiWebRadio". Entered the SSID and passphrase of my WLAN....and the connection was lost. means i could not find the device in my network. I tried to flash the software again but successless. 
Does someone know how i can start from scratch? i have already done a 'erase_flash' But nothing helps

thx
Ole

olemde, 2017-04-21:
Ok After several trials  i found that i have to Flash the Boot.bin again. Than it works! But  does someone know why  this is necessary?

karawin, 2017-04-21:
Have you entered a  IP, Mask and Gateway?
Have you checked or not the dhcp? 
Have you validated these infos?
etc...

And no. flash again the Boot is not a good solution.
Please try to find what you do wrong before any conclusions.

closed 2017-05-24 by karawin