karawin/Ka-Radio #29: Autoplay

1ron-pon3, 2017-05-11:
Hi! Is autoplay checkbox should enable autoplay last played station after booting up? If so, it doesn't work.

karawin, 2017-05-11:
Do not be so sure. 
It is necessary to check and uncheck just once the box to memorize it in karadio. 
After that, it's good forever.

closed 2017-05-11 by karawin

1ron-pon3, 2017-05-12:
I checked in unchecked it several times, tried to reboot end check it but the only result is the checkbox is sawed in web interface, but the autoplay itself doesn't work

karawin, 2017-05-12:
Which release?
Do an upgrade if not 1.2.1.

12 mai 2017 a écrit :
>I checked in unchecked it several times, tried to reboot end check it
>but the only result is the checkbox is sawed in web interface, but the
>autoplay itself doesn't work

-- 
Envoyé de mon téléphone Android avec <a href="https://play.google.com/store/apps/details?id=com.onegravity.k10.pro2"><b>K-@ Mail</b></a>. Excusez la brièveté.

1ron-pon3, 2017-05-12:
It's 1.2.1

karawin, 2017-05-12:
See the log on the serial usb.
Send it please

12 mai 2017 a écrit :
>It's 1.2.1

-- 
Envoyé de mon téléphone Android avec <a href="https://play.google.com/store/apps/details?id=com.onegravity.k10.pro2"><b>K-@ Mail</b></a>. Excusez la brièveté.

1ron-pon3, 2017-05-12:
It's funny, but from serial it looks like it works, but there is no sound before I press "Play" in web interface

`dhcp client start...
ip:192.168.0.104,mask:255.255.255.0,gw:192.168.0.1
autostart: playing:1, currentstation:0
##CLI.STOPPED# from playStationInt
##CLI.NAMESET#: 0 RM
##CLI.URLSET#: 93.125.53.192
##CLI.PATHSET#: /live
##CLI.PORTSET#: 8000
##CLI.OVOLSET#: 126
##CLI.VOL#: 254
##CLI.STOPPED# from Adc Stop
##CLI.ICY0#: Radio Maria Belarus
##CLI.ICY3#: http:\/\/www.radiomaria.by
##CLI.ICY4#: Religion
##CLI.ICY5#: 128
`

karawin, 2017-05-12:
#CLI.STOPPED# from Adc Stop

You forgot to wire adc (a0) input to the gnd

Le 12/05/2017 à 18:20, 1ron-pon3 a écrit :
> |dhcp client start...
> ip:192.168.0.104,mask:255.255.255.0,gw:192.168.0.1 autostart:
> playing:1, currentstation:0 ##CLI.STOPPED# from playStationInt
> ##CLI.NAMESET#: 0 RM ##CLI.URLSET#: [...]

1ron-pon3, 2017-05-12:
No, I didn't. It is wired. I even checked the connection with multimeter now, and the connection is definitely ok.

karawin, 2017-05-12:
So you have a problem with the adc input.
##CLI.STOPPED# from Adc Stop
The only reason to see this message is
			if (adc < 220) // stop
			{
				inside = true;
				clientDisconnect("Adc Stop");
			}

When the detection think that a control panel is wired.
if (system_adc_read() < 10) adcdiv = 0; // no panel adc grounded

Check directly on the esp8266 board
[image]
to see 0V?

1ron-pon3, 2017-05-12:
Oh, thank you very much. I understand now. The reason should be that I used NodeMCU Dev Kit, and it has a resistor on adc line between esp pin and connector. I will solder wire directly to adc pin now. Thank you again!

karawin, 2017-05-12:
The resistors are normal, they divide the input by 3.
If you don't read 0v when a0 is grouded, may be a failure on these resistors.
The software is able to see if it is a nodeMcu or a bare esp adc input.

1ron-pon3, 2017-05-15:
So, I have no idea what was the problem, but I solved it by wiring ADC to 3.3V with 16K resistor and to ground with 7k (like if there was a keyboard connected). Now it works. Thank you for your help again.