karawin/Ka-Radio #30: Playing resumes after pressing stop while power is on

romankot, 2017-05-19:
Steps to reproduce:
- start playing 
- stop by hardware button or  via webui 
- playing resumes after some time (from few seconds to several minutes) 
I expected no playing after pressing stop button. 
The issue is reproducible on v1.2.1

karawin, 2017-05-19:
Please Check adc to gnd connection.

karawin, 2017-05-19:
Sorry, not seen the hardware button...
Are your wires longs? May be an interference...

romankot, 2017-05-19:
Hardware button board connected with devkit with wires no more than 10cm long. Moreover volume buttons don't work every time correctly, next/previous works fine as well as stop/start

karawin, 2017-05-19:
This panel is very sensitive to the 3.3v accuracy and stability (and 
resistors precisions).
Check  to see if it is really 3.3 volts. The adc must be at 1.0 volt 
when no button in action

karawin, 2017-05-24:
No news?

closed 2017-05-24 by karawin