karawin/Ka-Radio #31: No reaction after flashing 1.3

hcremer, 2017-06-01:
Hi,
I tried to flash the new version on a ESP-12F. Did it before successfully with former version on other chips.

Now after flashing not getting access point and on  serial terminal @74880 baud I'm only getting:
" ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c "

On the same chip, I installed the WifiAccessPoint ESP8266Wifi sample scetch with the arduino ide and it worked fine as an access point.

Any ideas?

karawin, 2017-06-01:
It seemes that the boot is missing. See https://hackaday.io/project/11570-wifi-webradio-with-esp8266-and-vs1053

hcremer, 2017-06-01:
This is the command I used to flash:
sudo python2.7 esptool.py --port /dev/ttyUSB0 write_flash -ff 40m -fm qio -fs 32m 0x00000 boot_v1.6.bin 0x01000 user1.4096.new.4.bin 0x81000 user2.4096.new.4.bin 0x7e000 blank.bin 0x3fe000 blank.bin 0xfc000 esp_init_data_default.bin 0x3fc000 esp_init_data_default.bin

I also tried to flash under windows as shown in the screenshot https://cdn.instructables.com/FLS/H6IE/IZYGAMAW/FLSH6IEIZYGAMAW.SMALL.jpg?width=289.34070796460173

It always says that the flashing is successful. I tried the files downloaded from https://hackaday.io/project/11570-wifi-webradio-with-esp8266-and-vs1053 and also from the unzipped https://github.com/karawin/Ka-Radio.git

I also tried the 1.5 and the 1.6 boot files, but it never worked, ... (Arduino sample sketches for ESP8266 working fine on the same chip)

Do I maybe need to change fuses on these chips to load the bootloader @0x0000 when starting the chip?

karawin, 2017-06-01:
All seems correct. See the image of the right boot sequence.
[image]

hcremer, 2017-06-01:
I'm very confused now.
I tested it with another ESP-12F chip labeled: "MODEL: ESP8266MOD VENDOR: AI-THINKER"
Did exactly the same and it works fine.
The ones I bought at Bangood are labeled "MODEL: ESP8266MOD VENDOR: ESP-F" and told to be "Geekcreit® ESP-F ESP8266 Remote Serial Port WiFi IoT Module Nodemcu LUA RC Authenticity Compatible With ESP-12F" don't work with this strange behaviour described above.
What could be the difference (they both claim to be ESP-12F)?

karawin, 2017-06-01:
Don't know. May be a missing pullup on a gpio?

riedmueller, 2017-06-02:
Hello,
i have the same problem. I found out that i have to flash it with "DIO" SPI Mode 40MHz.

[image]

After flashing with this settings it seems to work (AP WifiWebRadio available, LED blinking)....

karawin, 2017-06-02:
So this model is not QIO. Probably to free the additional gpio's

hcremer, 2017-06-03:
Tried it with DIO
Now getting messages on serial@115200 baud but no access point viewable
[image]

karawin, 2017-06-03:
Something is blocking the vs1053 dialog. Check the dreq line.

hcremer, 2017-06-03:
After a few reboots it now works :-)
Thanks a lot to everybody for the help!

closed 2017-06-03 by hcremer