karawin/Ka-Radio #32: Resets and hangup's - strange behaviour

riedmueller, 2017-06-04:
Hallo,
i have built up a ka-radio out of an module VS1053B and an nodemcu V3.
I have flashed the nodemcu with firmware 1.3.
For now i never heared anything out of the VS1053 module (small handy earphone connected). I checked the connection and i think it's OK.

The web interface worked first for configuration, but now it hangs sometimes after:
- bootup
- Start playing a radio
- after rebootup by reset switch (example: hangup --> click reset switch --> reboot --> hangup -->click reset switch --> reboot --> hangup --> click reset switch --> now "overflow the heap_size"
Does a reset by reset switch not the same as powerup reset??
Only power off-on helps sometimes

Sometimes it shows strange behaviour in the settings:
- After false WIFI settings i could not reflash the system to get a clean system
  --> It seems that the settings are stored somewhere and stay consistent.
  --> Only way was to flash 1MB 0xFF dummy files 0n Adress 0x0000 0x10000 0x20000 0x30000
  --> After that reboot and reflash ka-radio
- Same behaviour with the station list. I wiped out everything with the 1MB 0xFF dummy files. But after flashing the firmware , the station list was not empty... So where was it saved? 
- Sometimes i get strange settings after reboot (but the settings are not changed before)
[image]

I attached a logfile after hangup and klicking 1 time (!!) on the reset button. 
[logfile.txt](https://github.com/karawin/Ka-Radio/files/1050509/logfile.txt)
It restarts several times and than hangs up.

For flashing i used this tool with DIO setting as QIO not works.
[image]

I have no logical conclusion because its a little bit confused problem.
So whats wrong here?
Could DIO instead of QIO be an problem?
Could it be that i have to change to another memory map 512+512 ---> 1024+1024 ?? (32Mbit-C1 ?)
Is there a precompiled testcode to check if the VS1053 is working?

Thanks!

karawin, 2017-06-04:
Hallo,
i have built up a ka-radio out of an module VS1053B and an nodemcu V3.

**Is it an AI-thinker ESP12E? ESP8266MOD?**

I have flashed the nodemcu with firmware 1.3.
For now i never heared anything out of the VS1053 module (small handy earphone connected). I checked the connection and i think it's OK.

The web interface worked first for configuration, but now it hangs sometimes after:
- bootup
- Start playing a radio
- after rebootup by reset switch (example: hangup --> click reset switch --> reboot --> hangup -->click reset switch --> reboot --> hangup --> click reset switch --> now "overflow the heap_size"
Does a reset by reset switch not the same as powerup reset??

**It is the same **

Only power off-on helps sometimes

**Check your power supply. Is it from usb?**

Sometimes it shows strange behaviour in the settings:
- After false WIFI settings i could not reflash the system to get a clean system

**If the AP is not good, it fails back to Webradio 192.168.4.1**

  --> It seems that the settings are stored somewhere and stay consistent.

**Yes in the end of the flash memory**

  --> Only way was to flash 1MB 0xFF dummy files 0n Adress 0x0000 0x10000 0x20000 0x30000
  --> After that reboot and reflash ka-radio
- Same behaviour with the station list. I wiped out everything with the 1MB 0xFF dummy files. But after flashing the firmware , the station list was not empty... So where was it saved? 

**A local mask is kept in the browser memory per ip**

- Sometimes i get strange settings after reboot (but the settings are not changed before)

I attached a logfile after hangup and klicking 1 time (!!) on the reset button. 
"A bit hard to read no? I cannot read it. full of strange chineese char.
Is it a 115200 bauds**

[logfile.txt](https://github.com/karawin/Ka-Radio/files/1050509/logfile.txt)
It restarts several times and than hangs up.

For flashing i used this tool with DIO setting as QIO not works.

**If QIO doesn't work and if it is an ESP12E something is physically broken in the flash memory access.**

I have no logical conclusion because its a little bit confused problem.
So whats wrong here?
Could DIO instead of QIO be an problem?

**In theory no.**

Could it be that i have to change to another memory map 512+512 ---> 1024+1024 ?? (32Mbit-C1 ?)

**No**

Is there a precompiled testcode to check if the VS1053 is working?

**No. with all these strange reactions, i am pretty sure that the power supply is not good.**

Thanks!

riedmueller, 2017-06-05:
Hi,
i bought it from al express. The picture of the seller shows AI-thinker on it. But my modele shows HYSIRY:
[image]
ANd here the VS1053B
[image]

I checked the power supply. It's from PC, but now i tested a 1A 5V USB Power supply without luck. Then i used laboratory supply 5V on VIN and GND of NodeMCU (VS1053 connected zto 5V of labsupply) also without better reactions.

Of corse the logfile is unreadble on some points, because of the 78800 baud. but the rest is readable w. notepad++...
[image]
....

My connection was
VS1053 --> NODEMCU
5V --> VIN
DGND --> GND
MOSI --> D7 (GPIO13)
DREQ --> D2 (GPIO4)
XCS --> D8 (GPIO15)
XDCS  --> D1 (GPIO5)
XRST --> D3 (GPIO0)
SCK --> D5 (GPIO14)
MISO --> D6 (GPIO12)
and on NODEMCU: VIN ==5V from lab-supply, A0==GND, EN on 3V

karawin, 2017-06-05:
All is good. It is not the first time that i see some problems with clone of ai-thinker.
Try these bin
http://karadio.karawin.fr/user1.4096.tmp.4.bin
http://karadio.karawin.fr/user2.4096.tmp.4.bin

The first log was really unreadable with the block note.

riedmueller, 2017-06-05:
Hello,
now  i disconnected all wires and flashed nodemcu firmware. it works for connectiong with lua interpreter - but only flashed with DIO.
After that i flashed your first ....tmp.4.bin. Switched power off and than one wire from A0-->GND.
After powerup via USB i get (78800):
[image]
Then i switched the terminal to 115200baud and make a reset via reset button:
[image]
Then i set the WIFI settings and the system makes an reboot. Her is the end of terminal output:
[image]
The low memory message comes some time after clicking through the pages in web interface without changing something. Is this message critical?
After repowering i can repeat this behaviour:
[image]
But: evering seems OK  with wifi connection and the behaviour is repeatable (reset button)

Now i connected the VS1053 module. This time powered with USB (measured 4.9V) VS1053 connected to VUSB.
Now the system hangs on wifi connect. It's not accessible... LED is not blinking. No answer on requests via uart interface.
[image]
I disconnected and connected usb three times with the same behaviour. Then i got a connect:
[image]

Then i used the web interface and restored the webstations list.
I start playing one radio...
[image]
Now the same again: 
- No audio aut of VS1053
- LED stops blinking
- No answer on uart requests

riedmueller, 2017-06-05:
And now i found out that if i disconnect D3 (GPIO0) during this hang up, the LED starts blinking again and connects sometimes to wifi...
And if i touch the insulation of this wire (XRST) an humming sound comes out of the earphone. Thats cool but not the sound that i want :- )

karawin, 2017-06-05:
Look at your log

[image]

Stopped from ADC stop means that a control switch is detected. Are you sur to have connected ADC to gnd?

I had another user with the same problem. The board was broken for adc. Check that the inpur of the esp is really at 0v

Send a log from the beginning with the vs1053 connected.

karawin, 2017-06-05:
Are you sure of your gateway?

riedmueller, 2017-06-05:
I think i have found the problem!!!

As described here:
[http://www.toughdev.com/content/2013/10/interfacing-vs1053-audio-encoderdecoder-module-with-pic-using-spi/](url)
> The real problem, after much research effort, is the XTEST pin on the VS1053. This pin must be connected to VDD for things to work consistently. Many clone boards, including earlier versions of the Sparkfun VS1053 boards and those sold on eBay, leave the XTEST pin unconnected, leading to unspecified behavior. I referred to the datasheet for the location of XTEST, which is pin 32 in the LQFP-48 package:

So i shorted PIN 32 with PIN31 and than it boots and works!!
 (and PIN34 with PIN33 because it starts in MP3 mode, but i think this was not the problem)

riedmueller, 2017-06-05:
And i checked ADC0 --> The input was directly connected to GND. Multimeter shows 0V. So i dont know why it says ...from ADC stop...
And now it works.
Could it be, that an missing or misconfigured VS1053 be a reason fro overflows, low memory or something else?

closed 2017-06-16 by karawin

Alnoor55, 2018-06-04:
Hi I got it working with the web interface, but unable to compile the karadioUCSTM32.ino , missing libary 
MapleFreeRTOS.h   . can you help please

karawin, 2018-06-04:
Read ucglibConf.h for the software required
Software:
---------
STM32DUINO  (to add support of the stm32 with arduino IDE:
https://github.com/rogerclarkmelbourne/Arduino_STM32
Bootloader (see http://wiki.stm32duino.com/index.php?title=Bootloader)
https://github.com/rogerclarkmelbourne/STM32duino-bootloader (the one we need is generic_boot20_pc13.bin)
Instruction to load the bootloader on stm32 (one time)
http://www.stm32duino.com/viewtopic.php?f=32&t=413

Libraries needed:
ucglib for stm32 must be installed in  arduino IDE library;
https://github.com/karawin/Ucglib_Arduino

IRMP library for stm32Duino:
https://github.com/karawin/irmp-master
See https://www.mikrocontroller.net/articles/IRMP_-_english

KY040 library:
https://github.com/karawin/encoder