karawin/Ka-Radio #34: No sound after 1.3.1 update

nlovrens, 2017-06-14:
Hi, I had my karadio working perfectly on v1.1.7. All i did was click on upgrade from web interface (didn't change or touch any other settings). Now I am running version 1.3.1 and everything looks like working but no sound is coming from VS1053 module.
Did something changed regarding hardware from v1.7.1 to v.1.3.1? I also have optional external RAM (23LCV1024) if that is relevant.
P.S. Thank you for all your great work regarding this project

karawin, 2017-06-14:
Try to disable the patch loading with uart command
sys.patch("0")

Some non vs1053b doesn't support the patch for aac even if it is detected as vs1053.

nlovrens, 2017-06-15:
Hi,
It works after disabling patch.
Thank you

karawin, 2017-06-15:
You are welcome

closed 2017-06-15 by karawin