karawin/Ka-Radio #35: Button Control - Problem since  newer Software

gerdfinder, 2017-07-04:
Hello Karawin,
Thanks for the great software but I now have a problem with newer software after OTA update.

The keys do not work as before.

I have an ESP-07 (32Mbit) so **no 2: 1 voltage divider**

With the newer software, the keys no longer function,
If I retrofit the 100K / 220K voltage divider **(like NodeMCU)**
the keys work as before.

I think my board is not recognized correctly.

Gerd

tomasf71, 2017-07-04:
Same for me :-(

karawin, 2017-07-04:
Yes, sorry I made a mistake.
Do an OTA again, I modified the release.

karawin, 2017-07-04:
Wrong correction. Must be ok now.
OTA again and tell me please.

arlaor, 2017-07-05:
Hi. @karawin, Are you planning to migrate your webradio to the esp32 module? To take advantage of the new processor and ram?
@Edzelf confirms that it improves the performance of the project.

tomasf71, 2017-07-05:
Still doesnt work for me :-(

gerdfinder, 2017-07-05:
Same for me.
Update yesterday and today (now 10:15 am)
not working without voltage divider.

karawin, 2017-07-05:
Yes sorry for this problem. I cannot test it here.
I found a wrong data on 8 bits in place of 16.
It may be good now.
OTA again.....

Thanks for your help.

karawin, 2017-07-05:
The esp32 is much better in term of speed and sram.
Yes, i plan to port karadio to it. A question of time, it is a lot of work even if karadio is already a freertos project.
The @Edzelf project is a pure arduino one. With a main loop and callbacks on esp8266. Not very good for a big software.
Karadio is yet a server with multi clients only limited by the sram. With the esp32 the limit is removed.
But for one or two clients the result will be quite the same in term of user feeling. Other limits  will be considered: the spi speed for example.

gerdfinder, 2017-07-05:
Thank you karawin, now it's working!

But the "Next" key works slow or not.

No keys pressed are exactly 1,000 V on the AD-in
"Next" pressed are there 0.895V, i think it's ok.

Gerd

gerdfinder, 2017-07-05:
I adjust the "No keys pressed are exactly 1,000 V on the AD-in"
to 0.980 V and all keys are working.

tomasf71, 2017-07-05:
Now it works as before, thank you

closed 2017-07-05 by gerdfinder

karawin, 2017-07-05:
I added an uart command
sys.adc
to show the value read on adc.
If you want to adjust some parameters from the following listing, let me know. I made these value may be too quickly.
	adc *= adcdiv;
		
	if ((adc >400) && (adc < 580)) // volume +
	{
		setVolumePlus();
	}
	else if ((adc >730) && (adc < 830)) // volume -
	{
		setVolumeMinus();
	}		
	if (!inside)
	{	
		if (adc < 220) // stop
		{
			inside = true;
			clientDisconnect("Adc Stop");
		}
		else if ((adc >278) && (adc < 380)) //start
		{
			inside = true;
			sprintf(Vol,"%d",currentStation);
			playStation	(Vol);
		}
		else if ((adc >830) && (adc < 920)) // station+
		{
			inside = true;
			wsStationNext();
		}
		else if ((adc >590) && (adc < 710)) // station-
		{
			inside = true;
		wsStationPrev();
		}
	}
}

arlaor, 2017-07-05:
Hi @karawin 
This is good news and I hope this migration will happen soon, because for some reason there are always intermittent cuts when listening to radio stations

karawin, 2017-07-05:
Intermittent cuts will not be resolved with the esp32.
They are due to the internet stream. If you can look at the instant bandwith used by your AP. You will see some hole or a saturation of your max bandwith. These hole if not too long are inexistant with the external memory. Yes, the esp32 will integrate this feature but it is not perfect if the 'hole' is more than few seconds.
Here i have a limited max speed of 400 kBytes and any download will cut the radio after 4 sec. on, off, on etc...
I don't see any QOS available on the esp nor my AP.

arlaor, 2017-07-09:
Hi @karawin
Sometimes I have to restart it up to 3 times to load correctly. But for what you tell me, it must be the internet. Thanks. I'll wait patiently for the version with esp32. In some way it will help in the performance

karawin, 2017-07-10:
Restat what? reset the nodemcu?
Check your power supply. It is abnormal.

arlaor, 2017-07-15:
Hi @karawin
I use 5V DC.  Is too much  voltage?

gerdfinder, 2017-07-15:
No, i use 230V AC,
but the Powerpin from ESP not more than 3V3!

karawin, 2017-07-15:
Vin 5V on nodeMCu is 5volt, 
Vcc on bare esp8266 is 3.3 V

arlaor, 2017-08-05:
Hi @karawin 
And when I connect the esp8266 module to my pc laptop to program it, it is receiving 5 volts, generated by the USB port. Should I use any voltage reducer to connect directly to the USB port on my laptop?