karawin/Ka-Radio #36: Web control for Volume not working in binary version 1.6

CaptainRoot, 2017-07-17:
Great Project! i used version 1.6 from the bin folder.
I noticed two things that i want you to know.
First, the volume control in the webinterface does not work. 
Second, i testet the restore station list function with your list, after that i was not able to hear any station. after i edited one station and set the volume offset to 126 everything works fine

Kind regards,
Stephan

karawin, 2017-07-17:
Thanks.
The current release is 1.3.4
May be you can try this one?
All is working. Only a bug on the web stop resolved in the next one.

CaptainRoot, 2017-07-17:
Hi, 
i used boot_v1.6.bin nd the binaries under Ka-Radio/ESP8266-Firmware/bin/upgrade/ so i guess it should be 1.3.4 the Webpage shows this in in upper right.
Volume Control is not working, i try it with Google Chrome

karawin, 2017-07-17:
Volume is working with a websocket on esp from the interface.
Hundred of users use it on this release.
For sure it works with all browser, edge chrome firefox etc.
The offset is for adjust the volume per station.

CaptainRoot, 2017-07-17:
Ok i retested ist wirh edge. works fine!
Google Chrome Version 59.0.3071.115 (Offizieller Build) (64-Bit) does not work ... but don't care 
Maybe an extension is destroying this feature ...

closed 2017-07-17 by CaptainRoot

karawin, 2017-07-18:
I test with Version 59.0.3071.115 (Build officiel) (64 bits) and it works. So may be as you said, an extension...