karawin/Ka-Radio #37: Volume Control via web interface not working

cad435, 2017-07-17:
Volume control via Web Interface not possible changing the Sliders makes no difference in Volume...
changing the Volume via UART (cli.vol+/cli.vol-) is possible and works just fine...

I get many socket errors:
Clear i: 0, socket: 2, errno: 107

maybe that have to do with it

=================================================
Startup Log:

=> extraram state: 0 0xff ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Release 1.3.4
SDK 1.4.2(f57d61a)
data  : 0x3ffe8000 ~ 0x3ffe899a, len: 2458
rodata: 0x3ffe89a0 ~ 0x3ffeb1dc, len: 10300
bss   : 0x3ffeb1e0 ~ 0x3fff2168, len: 28552
heap  : 0x3fff2168 ~ 0x40000000, len: 56984
Heap size: 29200
size_map: 4
t0 task: 3fff8fa0
t1 task: 3fff91b0
t4 task: 3fff96b8
t3 task: 3fff9d00
t2 task: 3fffa910
t5 task: 3fffaf58
Heap size: 19040
mode : sta(<MAC censored>)
add if0
UART READY
AP1: <SSID censored>, AP2: <SSID censored>
 AP1:Station Ip: <IP censored>
Trying <SSID censored> ,  I: 0 status: 1
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0x4C
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0xB800
I2S Speed: 0
plugin size 6350 3175 
plugin start: 7 1 8050 6
plugin size 234 117 
plugin start: 7 1 8f00 6
scandone
state: 0 -> 2 (b0)
Trying <SSID censored> ,  I: 1 status: 1
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 

connected with <SSID censored>, channel 12
dhcp client start...
ip:<IP censored>,mask:255.255.255.0,gw:<IP censored>
Clear i: 0, socket: 1, errno: 107
autostart: playing:1, currentstation:0
I2S Speed: 2
##CLI.STOPPED# from playStationInt
##CLI.NAMESET#: 0 #Musik-House
##CLI.URLSET#: house-high.rautemusik.fm
##CLI.PATHSET#: /
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 80
No panel
ADC Divisor: 0 from adc: 0
Header: Moved
##CLI.STOPPED# from C_HEADER
##CLI.URLSET#: de-hz-fal-stream07.rautemusik.fm
##CLI.PATHSET#: /house
##CLI.PORTSET#: 80
##CLI.ICY0#: #MUSIK.HOUSE (FUNKY) - WWW.RAUTEMUSIK.FM - 24H MIXED HOUSE DANCE ELECTRO FUNK MINIMAL AND MORE!
##CLI.ICY1#: <BR>This stream requires <a href=\"http:\/\/www.winamp.com\/\">Winamp<\/a><BR>
##CLI.ICY2#: SHOUTcast Distributed Network Audio Server\/Linux v1.9.8<BR>
##CLI.ICY3#: http:\/\/www.rautemusik.fm\/house\/
Clear i: 0, socket: 2, errno: 107
##CLI.ICY4#: House Electro Electronic
##CLI.ICY5#: 192
##CLI.PLAYING#
##CLI.META#: Shirox - Bass In Your Face
Clear i: 0, socket: 2, errno: 107

karawin, 2017-07-18:
Yes it seems that the socket is closed when karadio try to write to it. 
 No idea why.
 browser? This socket is opened by the browser.

cad435, 2017-07-18:
I use Google Chrome on Windows 7 64-Bit, but can try it with FF on Windows and Chrome on Android later at home... 

all other controls are working via web interface, only Volume is not...

karawin, 2017-07-18:
It seems that some network administrator traps the websocket....
I will revert to a standart POST for volume.
All other websocket flow will remains the same (meta display etc....)

cad435, 2017-07-18:
ok, I tested it:

Chrome on Android 6.0.1 (CM13.1.2): nope
FF 54.0.1(32-Bit release) on Windows 7 64-bit: it works indeed...

closed 2017-07-19 by karawin