karawin/Ka-Radio #41: No AP point after flash,

Edragon, 2017-07-27:


karawin, 2017-07-27:
The vs1053 acces is blocking. 
Check the DREQ line or disconnect it to be sure it is the problem.

Edragon, 2017-07-27:
Yes, indeed, all good, serial output correctly. probably some soldering problem, when I have vs1053 all work. 

Thank you very much.

Edragon, 2017-07-27:
Hi Karawin,

Yes one more question, there is a wiring instruction in the readme.md:
From ESP8266_ESP12( 3.3 v) to VS1053 (5 v)
REST
ADC if control panel is not used, this input must be grounded.
CH_PD to 3.3v
GPIO16 (a 1Hz output)
GPIO14 to VS1053 SCK
GPIO12 to VS1053 MISO
GPIO13 to VS1053 MOSI

TXD to CH340G UART rx
RXD from CH340G UART tx
GPIO05 to VS1053 XDCS
GPIO04 to VS1053 DREQ
GPIO00 to VS1053 XRST
GPIO02
GPIO15 to VS1053 XCS

But there is also a schematic called webradio in hardware folder, should I use the readme.md wiring?  Thanks!

Edragon, 2017-07-27:
in the schematic pdf, io5 is connected to DREG, I am not sure if it is the wiring problem.

karawin, 2017-07-27:
This schematic is the old one from piotr. Don't use it.
DREQ is on GPIO04 (d2)

Edragon, 2017-07-27:
good to know, thanks!

closed 2017-07-27 by Edragon