karawin/Ka-Radio #42: Support Cyrillic characters required for channel/song name

romankot, 2017-07-28:


karawin, 2017-07-28:
Only by the browser. not by serial or telnet.

romankot, 2017-07-30:
Terminal doesn't show Cyrillic fonts. Should I change remote character set on putty to see Cyrillic ?

karawin, 2017-07-30:
I don't know. Only the basic char stream is sent.
See it on a browser.
[image]

romankot, 2017-08-02:
you see 

[image]

karawin, 2017-08-02:
This station http://uk5.internet-radio.com:8112/live.m3u is not in cyrillic and it seems that the browser don't know the translation if possible.

karawin, 2017-08-08:
Cannot do something for that.

arlaor, 2017-08-11:
Hi @karawin 

Until version 1.4.2 I was able to update the project, but with versions 1.5 and 1.5.1 does not take the changes and is maintained in version 1.4.2.
Can you add a warning next to the upgrade button, which lets you know if the project is being updated or not?
I have done the same steps for the previous updates but in version 1.5 has not taken the changes.
Thanks

karawin, 2017-08-11:
On the serial interface or telnet do a sys.update
if unsuccesfull, it is clearly written.
In this case do a sys.prerelease
Things must be right now.

arlaor, 2017-08-11:
The system Prerelease Should I do it for some special program?

karawin, 2017-08-11:
No, it is the current one for the moment. I said that because it is possible that your 1.4.2 points to a test station. (a wrong release at a moment). But try the normal update, i may have corrected it by placing a copy of the good bin.

closed 2017-08-25 by karawin