karawin/Ka-Radio #44: After receive data to STM32 problem

ogggi, 2017-07-30:
Hello

I have problem after turning on karadio. ESP(Nodemcu) normal playing, but STM32F103(with LCD ST7735,encoder and IR) after receive signal from serial ESP, itself stop.  Led diode on STM32 not blinking,but shines and lcd it stop. 
Sometimes normal working and all is good.

thank you for answer

karawin, 2017-07-30:
Is it the soft from https://github.com/karawin/karadio-addons ?
Not sure that it is a software problem.
If on a breadboard, check all lines and may be double the gnd and vcc.
I know that many breadboards are not very accurate (the long white from china).

ogggi, 2017-07-31:
I havent on breadboard. All wires are soldering on board.  I have one question.  How you use library freertos with stm32?

ogggi, 2017-07-31:
when I get reset on stm32 and receive data from esp ,stm32 freeze. I have two pieces stm32 and nodemcu, on both doing same.

karawin, 2017-07-31:
I use it everyday on two prototypes.
I delivered again with 1.3 release.
May I forgot to update the github.
Please reload.

karawin, 2017-07-31:
This issue is for addon, not for KaraDio itself. 
Please move it.

closed 2017-07-31 by karawin