karawin/Ka-Radio #45: telenet sys.version doesn't work on 1.4.2

romankot, 2017-08-01:
Could not find the file with all command list (interfaces.txt) on github

karawin, 2017-08-01:
[image]
[image]
[image]

karawin, 2017-08-01:
type 
help
or see the setting panel or directly http://karadio.karawin.fr/Interface.txt

romankot, 2017-08-02:
I see nothing from sys.version. Setting on putty set exactly as you show. sys.version is not described in interfaces.txt

karawin, 2017-08-02:
Telnet negociation mode passive too?

romankot, 2017-08-02:
yes

karawin, 2017-08-02:
This is a new command. Be sure to be with 1.4.1 or 1.4.2 release.

romankot, 2017-08-03:
it works ! to be close

closed 2017-08-03 by romankot