karawin/Ka-Radio #46: Radio hangup after several telnet "cli.vol-" commands in a row

romankot, 2017-08-02:


karawin, 2017-08-02:
in a row????

romankot, 2017-08-02:
try send several vol- one after one within 2 seconds

karawin, 2017-08-02:
Not reproductive

romankot, 2017-08-03:
Jean, try repeat vol- fast enough with upwards arrow. It seems that telnet get overloaded with fast input and drop off session. The radio itself keeps working but telnet session must be restarted

karawin, 2017-08-03:
Not reproductive.
Tested many time. But...
Is it a telnet client problem?

romankot, 2017-08-03:
i use putty

romankot, 2017-08-06:
Jean, do you filter number of incoming web requests? Radio hung ups after several fast start/stop combination

karawin, 2017-08-08:
May be filtered in case of extreme load. But in many case, just delayed.
Try the 1.5 release please.

romankot, 2017-08-08:
not reproducible on 1.5 anymore. to be closed

closed 2017-08-08 by romankot