karawin/Ka-Radio #47: Wrong VS1053 detect

Edragon, 2017-08-08:
Hi!

The output debug info show the VS1053 detect =0, but should be 4 right? I have tried two VS1053 boards, all the same, my esp8266 board is wemos D1. Please help. thanks!

=> extraram state: 0 0xff 
Release 1.4.2, Revision 3
SDK 1.4.2(f57d61a)
data  : 0x3ffe8000 ~ 0x3ffe8930, len: 2352
rodata: 0x3ffe8930 ~ 0x3ffea4a4, len: 7028
bss   : 0x3ffea4a8 ~ 0x3fff18b0, len: 29704
heap  : 0x3fff18b0 ~ 0x40000000, len: 59216
Heap size: 30304
size_map: 4
t0 task: 3fff8b48
t1 task: 3fff8dd0
t4 task: 3fff93c8
t3 task: 3fff97b8
t2 task: 3fff9db0
t5 task: 3fffa1c8
Heap size: 22712
mode : sta(2c:3a:e8:0a:ed:b1)
add if0
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0xC
VS Version (VS1053 is 4) = 0
SCI_ClockF = 0xB000
device: 3fffabe8
volume: 0
UART READYAP1: sscz_48FD, AP2: 
 AP1:Station Ip: 192.168.0.100
Trying sscz_48FD ,  I: 0 status: 1
scandone
state: 0 -> 2 (b0)
Trying sscz_48FD ,  I: 1 status: 1
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

connected with sscz_48FD, channel 7
dhcp client start...
ip:192.168.0.104,mask:255.255.255.0,gw:192.168.0.1
autostart: playing:0, currentstation:0
sleep enable,type: 2
No panelADC Div: 0 from adc: 2

Edragon, 2017-08-08:
the esp8266 logic should be 3.3V, vs1053 board power is 5V, is there any logic conflict?

karawin, 2017-08-08:
A good wired vs1053b shows
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0x4C
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0xB800

According to the datasheet:
SS_VER is 0 for VS1001, 1 for VS1011, 2 for VS1002, 3 for VS1003, 4 for VS1053 and VS8053, 5 for VS1033, 7 for VS1103, and 6 for VS1063.
Are you sure that you have a vs1053?

karawin, 2017-08-08:
The vs1053 power is 5v but an internal reducer is a 5v to 3.3v to power the chip.

Edragon, 2017-08-08:
I have tried a 3.3V power supplied and logic VS1053 board, connect directly to esp8266, but same error. Then I make the second test with the board you mentioned here:
http://www.ebay.fr/itm/401046111343?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

Same board but still the same error, so I guess it is a problem of my wemos D1 board. But this board should be very similar to nodemcu board you used .. Both nodemcu and wemos board has reset circuit on GPIO pin0, does it make problem?  Will try different board later and report result. Thanks!

karawin, 2017-08-08:
The vs1053 MUST be powered at 5V.

karawin, 2017-08-08:
Some uses the wemos D1 succesfully.

Edragon, 2017-08-08:
Got the issue solved, stupid :) wiring io4 and io5 swapped and read wrong.

Thanks for all the support!

closed 2017-08-08 by Edragon