karawin/Ka-Radio #48: Playing a .wav file

alpbak, 2017-08-22:
Hi,
First of all thank you for all your efforts.

I have no problems with mp3 playing and internet radios, everything works just perfect for me. But I would like to play wav files in Ka-Radio and I seem to have a problem with that.

What I'm trying is to enter the following wav file URL into the Instant Play section of the web interface:

[http://www.wavsource.com/snds_2017-07-30_6786517629734627/movie_stars/bogart/dirty_mind.wav](http://www.wavsource.com/snds_2017-07-30_6786517629734627/movie_stars/bogart/dirty_mind.wav)

The first time I play, it plays the file. But it doesn't play the file a second time and it doesn't play any other wav files anymore. 

When I restarted the system, it again plays a single wav file and stop playing any more files.

The Serial Monitor output is here:

dhcp client start...
ip:192.168.43.199,mask:255.255.255.0,gw:192.168.43.1
sleep enable,type: 2
ADC Div: 3 from adc: 13
autostart: playing:0, currentstation:0
##CLI.STOPPED# from Adc Stop
##CLI.STOPPED# from Post instant_play
##CLI.URLSET#: www.wavsource.com
##CLI.PATHSET#: /snds_2017-07-30_6786517629734627/movie_stars/bogart/dirty_mind.wav
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 160
##CLI.PLAYING# (This one is playing correctly)

##CLI.STOPPED# from once
##CLI.STOPPED# from Post instant_play
##CLI.URLSET#: www.wavsource.com
##CLI.PATHSET#: /snds_2017-07-30_6786517629734627/movie_stars/bogart/dirty_mind.wav
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 160
##CLI.PLAYING# (This one is not playing correctly even though it's the same file)

##CLI.STOPPED# from once
##CLI.STOPPED# from Post instant_play
##CLI.URLSET#: www.wavsource.com
##CLI.PATHSET#: /snds_2017-07-30_6786517629734627/movie_stars/bogart/need_dough.wav
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 160
##CLI.PLAYING# (This one is not playing correctly and this is another file)

##CLI.STOPPED# from once

Can you please recommend a solution to me or show me a hint where I can hunt down this issue.

Thanks in Advance
Alp

alpbak, 2017-08-22:
This was an internet connection issue and nothing to do with Ka-Radio. Sorry.

alpbak, 2017-08-22:
It looks like this is not an internet connection issue after hours of testing the issue still remains a mystery.

karawin, 2017-08-22:
On a telnet session or on serial type
sys.prerelease

It is an issue with the current release, i think.

alpbak, 2017-08-22:
It's on a serial type and I use the web interface to use the instant play feature.

I've stumbled upon a stranger issue while trying to solve this:

I can play this file only one time and Ka-Radio doesn't play this file again.
http://www.wavsource.com/snds_2017-07-30_6786517629734627/movie_stars/bogart/dirty_mind.wav

But if I try the following file, I can play it as many times as I want, back to back, with no problems as it's expected to be.
http://www.wavsource.com/snds_2017-07-30_6786517629734627/movie_stars/bogart/need_dough.wav

Any idea why this may be happening?

karawin, 2017-08-22:
It is the vs1053 problem.
Update to the future release with
sys.prerelease
command on the serial input.

alpbak, 2017-08-22:
I did the "sys.prerelease" and now it is "Release 1.5.3, Revision 1 SDK 1.4.2(f57d61a)" but nothing has changed the same problem continues, unfortunately. 

But the real interesting point is the issue I described in my previous message that a wav file is played correctly whereas the other one causes the problem. 

I think I'm stuck at this end :(

arlaor, 2017-09-13:
Hi @karawin 
you can integrate the 3 options (Radio, edit, and settings) into the fixed part of the header.
When I am at the bottom of the page, I should always go back to the beginning of the page to search for these options, to navigate to another option.
Thanks

karawin, 2017-09-15:
Thanks for your suggestion.
Yes it bother me too.
But for small screen it is not a good solution.
I propose a click on the header to scroll to the top. What do you think?

arlaor, 2017-09-15:
Hi @karawin 

it is a great idea to leave the main header fixed and allow the three options to be displayed with a click
thanks

karawin, 2017-11-10:
I think i can close. Reopen if needed.

closed 2017-11-10 by karawin