karawin/Ka-Radio #50: time.nist.gov is wrongly attributed in lots of wolrdwide DNS servers

j-levy, 2017-11-10:
I noticed that the domain name time.nist.gov used in ntp.c is most of the time wrongly attributed in most of the free DNS I've tested in my country (The Netherlands), as well as in worldwide public DNS such as Google's (**8.8.8.8 / 8.8.4.4**) or OpenDNS (**208.67.222.222 / 208.67.220.220**). Nearly all the time, they answer a wrong IP which isn't reachable, but it's not deterministic, on exceptional cases it works.
By switching for some German DNS (from Chaos Computer Club : **194.150.168.168** ) in my router, all the requests are now done correctly (tested with PING).
I am aware that it's a very personal problem (it seems that most, if not all Dutch DNS are affected, I didn't check all of them in that list : https://public-dns.info/nameserver/nl.html , and some are also affected in Germany), but some way to fix this would be : 
- either putting a fixed IP in **ntp.c** for the Karadio to ping all the time the same NTP server, regardless of the DNS chain between them,
- or putting a specific domain name for a specific DNS (for example, time-c-g.nist.gov )
- or allowing to enter a custom NTP server on the web page.

Solutions 1 or 2 are probably the easiest to implement, and I highly doubt that some tens of karadios over the world would take an NTP server down (that's my opinion though).

Thank you for all your hard work !

karawin, 2017-11-10:
Right it is a problem. For esp32 I switched to pool.ntp.org and it seems much better.
I will change it too for the esp8266.
But you are right, i will think to add a command to specify the hst to use (name or ip)

karawin, 2017-11-10:
Type sys.prerelease to update and tell me...

j-levy, 2017-11-10:
After launching the update with sys.prerelease, and re-putting my old (and faulty) DNS, it seems to be working much better. Thank you for the update !

closed 2017-12-19 by karawin