karawin/Ka-Radio #54: No AccessPoint after flashing

gvoigtlaender, 2017-12-22:
Hey there,

Getting no AP after flashing as well, unfortunately not gone after several reboots, as well as double-checking all connections did not show up any obvious faults

I used Dimitris Saltas PCB [1] manufactured by linked OSH Park [2]. Only obvious difference is I got an ESP-12F which is not AI-Thinker [3]. So far I assembled two boards, both show same result:

> External ram detected
>uart speed: 115200
>Release 1.6.2, Revision 2
>SDK 1.4.2(f57d61a)
>data  : 0x3ffe8000 ~ 0x3ffe88fc, len: 2300
>rodata: 0x3ffe8900 ~ 0x3ffea2f4, len: 6644
>bss   : 0x3ffea2f8 ~ 0x3fff1718, len: 29728
>heap  : 0x3fff1718 ~ 0x40000000, len: 59624
>Heap size: 44800
>size_map: 4
>Eeprom_start: 3e0000
>Eeprom_start1: 3d0000
>Flash size: 4194304
>t0 task: 3fff52a8
>t1 task: 3fff5530
>t2 task: 3fff5b50
>t3 task: 3fff5f40
>t4 task: 3fff6588
> Heap size: 38016
> mode : sta(2c:3a:e8:17:74:df)
> add if0

I already tried flashing in QIO as well as DIO mode - both come up with same result.

Is there any difference between my ESP and the AI-Thinker one? 
Other issues suggested disconnecting the DREQ line between ESP-12F & VS1053B, but there were no more information about if that did anything. I'm holding off that attempt as it could easily break the soldering pads.
Any suggestion what else to check? 

Thanks for any help,
georg

[1] https://github.com/dsaltas/WiFi-WebRadio 
[2] https://oshpark.com/shared_projects/H0BeSd2u
[3] https://de.aliexpress.com/item/Free-Shipping-10pcs-lot-ESP8266-remote-serial-Port-WIFI-wireless-module-through-walls-Wang-esp-12F/32471895953.html?spm=a2g0s.9042311.0.0.sL94xJ

karawin, 2017-12-28:
Please give the complete log.

gvoigtlaender, 2017-12-28:
I double-checked all connections from ESP to VS, I also confired no cross-bridges between other lines of VS.

Flash binaries:
>georgv@xps ~/esp/karadio-flash $ ls -la
>insgesamt 764
>drwxr-xr-x  2 georgv users   4096 28. Dez 15:00 .
>drwxr-xr-x 20 georgv users   4096 28. Dez 15:00 ..
>-rw-r--r--  1 georgv users   4096 15. Dez 12:27 blank.bin
>-rw-r--r--  1 georgv users   4080 15. Dez 12:27 boot_v1.7.bin
>-rw-r--r--  1 georgv users    128 15. Dez 12:27 esp_init_data_default.bin
>-rw-r--r--  1 georgv users 377444 15. Dez 12:27 user1.4096.new.4.bin
>-rw-r--r--  1 georgv users 377444 15. Dez 12:27 user2.4096.new.4.bin

Flashing log:
>georgv@xps ~/esp/karadio-flash $  esptool.py write_flash 0x000 boot_v1.7.bin 0x01000 user1.4096.new.4.bin 0x81000 user2.4096.new.4.bin 0x7e000 blank.bin 0x3fe000 blank.bin 0xfc000 esp_init_data_default.bin 0x3fc000 esp_init_data_default.bin
>esptool.py v2.2
>Connecting...
>Detecting chip type... ESP8266
>Chip is ESP8266EX
>Uploading stub...
>Running stub...
>Stub running...
>Configuring flash size...
>Auto-detected Flash size: 4MB
>Flash params set to 0x0040
>Compressed 4080 bytes to 2936...
>Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 123.7 kbit/s)...
>Hash of data verified.
>Compressed 377444 bytes to 277664...
>Wrote 377444 bytes (277664 compressed) at 0x00001000 in 24.5 seconds (effective 123.2 kbit/s)...
>Hash of data verified.
>Compressed 377444 bytes to 277696...
>Wrote 377444 bytes (277696 compressed) at 0x00081000 in 24.5 seconds (effective 123.4 kbit/s)...
>Hash of data verified.
>Compressed 4096 bytes to 26...
>Wrote 4096 bytes (26 compressed) at 0x0007e000 in 0.0 seconds (effective 4063.5 kbit/s)...
>Compressed 4096 bytes to 26...
>Wrote 4096 bytes (26 compressed) at 0x003fe000 in 0.0 seconds (effective 4108.4 kbit/s)...
>Hash of data verified.
>Hash of data verified.
>Compressed 128 bytes to 75...
>Wrote 128 bytes (75 compressed) at 0x000fc000 in 0.0 seconds (effective 82.0 kbit/s)...
>Hash of data verified.
>Compressed 128 bytes to 75...
>Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 82.0 kbit/s)...
>Hash of data verified.
>
>Leaving...
>Hard resetting...
>georgv@xps ~/esp/karadio-flash $ 

Debug output, started before reset by disconnecting power to pcb
>georgv@xps ~/esp/karadio-flash $ screen /dev/ttyUSB0 115200
>
> --- lot of empty lines
> 
>{ll��|�d�|�d�b|����s�b�#��no�$g'���cp��dsdsdx�o��d#g�|�$�|�#��no�$��l`�gl`'{���g�{l x�n�{�����co�|쏇�c��o'�l �n�lorǛ�'�d`8�g�r������p8'�|ć��c��'o��d`�nd`or���nc�d`{��ob�l`sd��g���og����s'o�l�c�br��䇎d{$��l������lllrl��$$dn���d�$l$��l��o�g�'��dl�c��l$xc�cs���l gd�p��d`bl`l��o�'���gd�|
>
> --- lot of empty lines
> 
>dl`{ls��c���#$�b�#`��;���x��n�og��l�xl�$��ll��d��l �o�ğ��
>External ram detected
>uart speed: 115200
>Release 1.6.2, Revision 2
>SDK 1.4.2(f57d61a)
>data  : 0x3ffe8000 ~ 0x3ffe88fc, len: 2300
>rodata: 0x3ffe8900 ~ 0x3ffea2f4, len: 6644
>bss   : 0x3ffea2f8 ~ 0x3fff1718, len: 29728
>heap  : 0x3fff1718 ~ 0x40000000, len: 59624
>Heap size: 44800
>size_map: 4
>Eeprom_start: 3e0000
>Eeprom_start1: 3d0000
>Flash size: 4194304
>t0 task: 3fff52a8
>t1 task: 3fff5530
>t2 task: 3fff5b50
>t3 task: 3fff5f40
>t4 task: 3fff6588
>Heap size: 38016
>mode : sta(2c:3a:e8:17:74:df)
>add if0

Any other logs I could provide?

karawin, 2017-12-28:
A normal log will show:
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0xC
VS Version (VS1053 is 4) = 0
SCI_ClockF = 0xB000
device: 3fffabe8
volume: 0

after the addi f0
If blocked here, the only answer is that the dreq line never go to 1 or 0 (don't remind) and the software loop.
I know that without vs1053, the soft continue (but no sound obviously)
Something is wrong with your vs1053.
Take a snap of the board.

karawin, 2017-12-29:
Test the dreq line connected to gpio4. It must be at level 1 3.3v)

gvoigtlaender, 2017-12-30:
Line dreq (gpio4) is 0V. Need to pull-up? I'm not so firm with electronic details ;(

I ordered material for another 3 boards which I'll assemble with AI-Thinker ESPs, going to check if they work

karawin, 2017-12-30:
At power up, the vs1053 pulls the dreq line, so I think the vs1053's weld on the dimitris board fails. Look at it with a magnifying glass and you'll be scared

gvoigtlaender, 2017-12-30:
Unfortunately while looking absolute beginner-like all welds are working - wiring test looks good, no cross-connection between adjoining legs, lines meet their counterparts. That goes for both boards I assembled. 

What is working: flashing & bootup ESP, so that is at least not totally burned. When powered up I can confirm both: 3.3V & 1.8V at all vias. XDCS (GPIO5), XRESET (GPIO0), LED (GPIO2), XCS (GPIO15), CS(?) (GPIO16) are 3.3V

I'll go back to the breadboard with a ESP12-adapter-board, lets see if i can get it running without node-mcu

gvoigtlaender, 2017-12-30:
OK, on breadboard both ESPs (AI-Thinker & unknown clone) work perfect with LC Technology VS1053 

Now i'll try to go on with VS1053b on lqpf48 adapter board

gvoigtlaender, 2017-12-30:
That did also not work. I only connected the supplies (GND, 3.3V@Pin32) and communication lines (MISO, MOSI, SCK, XCS, XDCS, DREQ, XRESET), still stopping after "add if0" ;(

Do I need to connect the oscillator and pull-down the GPIO{0-7}? Thought that part is not required for initial communication...

karawin, 2017-12-30:
Look at the scheme https://github.com/dsaltas/WiFi-WebRadio/blob/master/WiFiRadio.pdf
The vs1053 needs all parts.
If the dreq line is not high, the software cannot run and loop waiting dreq.

gvoigtlaender, 2017-12-30:
OK, that's gonna take a while, need to get all parts as through-hole components.

I'm going to file my mouser order list later, maybe someone will find some failure in there

gvoigtlaender, 2017-12-30:
Here is the order list of mouser, maybe someone has time to confirm against https://github.com/dsaltas/WiFi-WebRadio/blob/master/BOM.pdf , same sequence, without EPS & VS1053

Mouser No: 667-ERJ-6ENF1202V 
Mfr. No: ERJ-6ENF1202V 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 0805 12Kohms 1% AEC-Q200 

Mouser No: 652-CR0805FX-1004ELF 
Mfr. No: CR0805-FX-1004ELF 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 1M 1% 

Mouser No: 603-RC0805FR-07100KL 
Mfr. No: RC0805FR-07100KL 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 100K OHM 1% 

Mouser No: 603-RC0805FR-07470RL 
Mfr. No: RC0805FR-07470RL 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 470 OHM 1% 

Mouser No: 652-CR0805FX-4702ELF 
Mfr. No: CR0805-FX-4702ELF 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 47K 1% 

Mouser No: 603-RC0805FR-0710KL 
Mfr. No: RC0805FR-0710KL 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 10K OHM 1% 

Mouser No: 603-RC0805FR-0720KL 
Mfr. No: RC0805FR-0720KL 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 20K OHM 1% 

Mouser No: 652-CR0805FX-20R0ELF 
Mfr. No: CR0805-FX-20R0ELF 
Desc.: 
Thick Film Resistors - SMD Thick Film Resistors - SMD 20ohm 1% 

Mouser No: 81-GRM21BC81C106KE5K 
Mfr. No: GRM21BC81C106KE15K 
Desc.: 
Multilayer Ceramic Capacitors MLCC - SMD/SMT Multilayer Ceramic Capacitors MLCC - SMD/SMT 0805 10uF 16volts *Derate Voltage/Temp 

Mouser No: 80-C0603C104K5R 
Mfr. No: C0603C104K5RACTU 
Desc.: 
Multilayer Ceramic Capacitors MLCC - SMD/SMT Multilayer Ceramic Capacitors MLCC - SMD/SMT 50volts 0.1uF X7R 10% 

Mouser No: 80-C0805C332K5RAC 
Mfr. No: C0805C332K5RAC 
Desc.: 
Multilayer Ceramic Capacitors MLCC - SMD/SMT Multilayer Ceramic Capacitors MLCC - SMD/SMT 50volts 3300pF 10% X7R 

Mouser No: 647-UKA1C221MED 
Mfr. No: UKA1C221MED 
Desc.: 
Aluminium Electrolytic Capacitors - Leaded Aluminium Electrolytic Capacitors - Leaded 16Volts 220uF 20% Tol. 

Mouser No: 80-C0805C220J5G 
Mfr. No: C0805C220J5GACTU 
Desc.: 
Multilayer Ceramic Capacitors MLCC - SMD/SMT Multilayer Ceramic Capacitors MLCC - SMD/SMT 50volts 22pF C0G 5% 

Mouser No: 81-GRM40X103K50D 
Mfr. No: GRM216R71H103KA01D 
Desc.: 
Multilayer Ceramic Capacitors MLCC - SMD/SMT Multilayer Ceramic Capacitors MLCC - SMD/SMT 0805 0.01uF 50volts X7R 10% 

Mouser No: 81-GRM21R61C105KA01L 
Mfr. No: GRM21BR61C105KA01L 
Desc.: 
Multilayer Ceramic Capacitors MLCC - SMD/SMT Multilayer Ceramic Capacitors MLCC - SMD/SMT 0805 1uF 16volts X5R 10% 

Mouser No: 81-BLM18PG221SN1D 
Mfr. No: BLM18PG221SN1D 
Desc.: 
Ferrite Beads Ferrite Beads 220ohms 

Mouser No: 579-23LC1024-I/SN 
Mfr. No: 23LC1024-I/SN 
Desc.: 
SRAM SRAM 1024K 2.5V SPI SERIAL SRAM SQI 

Mouser No: 579-MCP1703AT1802ECB 
Mfr. No: MCP1703AT-1802E/CB 
Desc.: 
LDO Voltage Regulators LDO Voltage Regulators LOW Iq, High Vin LDO 1.8V, SOT-23A 

Mouser No: 579-MCP1703AT3302ECB 
Mfr. No: MCP1703AT-3302E/CB 
Desc.: 
LDO Voltage Regulators LDO Voltage Regulators LOW Iq, High Vin LDO 3.3V, SOT-23A 

Mouser No: 815-ABL-12.288-B2 
Mfr. No: ABL-12.288MHZ-B2 
Desc.: 
Crystals Crystals +/-20ppm 12.288MHz FUNDAMENTAL 

Mouser No: 688-SKQGAD 
Mfr. No: SKQGADE010 
Desc.: 
Tactile Switches Tactile Switches 5.2x5.2x1.5mm 260gf 

Mouser No: 806-STX-3100-5N 
Mfr. No: STX-3100-5N 
Desc.: 
Phone Connectors Phone Connectors 3.5mm PCB STEREO BLK 5P W/SWITCH

closed 2019-02-16 by karawin