karawin/Ka-Radio #55: edit several stations radio fail

arlaor, 2017-12-24:
Hi @karawin 
when trying to edit several radio stations, the page is disconnected from the project and I lose control.
Please allow support for stations of the type:
http://hot.friezy.ru/?radio=di&station=bassnjackinhouse&bitrate=128

Thanks

karawin, 2017-12-25:
What is the problem with the radio?
Bass & Jackin' House - Digitally Imported Premium   
It works well here.

And the edit don't disconnect when i test it for a lot of radio.

arlaor, 2017-12-27:
Hi @karawin 

The problem is this: when I play the station from INSTANT PLAY, I have no problem, but when I load the list of stations with the station stored in the list, the project shows the following information

URL STATION on INSTANT PLAY
Header: Moved
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: prem2.di.fm
##CLI.PATHSET#: /bigroomhouse?d97b84c0668de93f41871fb3
##CLI.PORTSET#: 80
##CLI.ICY0#: Big Room House - Digitally Imported Premium
##CLI.ICY1#: <BR>This stream requires <a href=\"http:\/\/www.winamp.com\/\">Winamp<\/a><BR>
##CLI.ICY2#: SHOUTcast Distributed Network Audio Server\/Linux v1.9.8<BR>
##CLI.ICY3#: http:\/\/www.di.fm
##CLI.ICY4#: House Big Tech
##CLI.ICY5#: 128
##CLI.PLAYING#
AND...
URL FROM LIST SAVED AND LOAD
##CLI.META#: David Guetta Feat. Nicki Minaj - Turn Me On (Michael Calfan Remix)
Warning eebuf malloc low memory
##CLI.STOPPED# from playStationInt
##CLI.NAMESET#: 166 CLUBHITS DI.FM
##CLI.URLSET#: hot.friezy.ru
##CLI.PATHSET#: /?radio=di
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 98
After this, it does not respond until the reset button is pressed 

please generate list webradio for probe
{"Name":"00srock (RR) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rr&station=00srock&bitrate=128","Port":"80","ovol":"0"}
{"Name":"80 HotMixRadio (80's) mp3 128 kbps","URL":"streaming.hotmix-radio.net","File":"/hotmixradio-80-128.mp3","Port":"80","ovol":"0"}
{"Name":"80srock (RR) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rr&station=80srock&bitrate=128","Port":"80","ovol":"0"}
{"Name":"80srock (RT) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rt&station=80srock&bitrate=128","Port":"80","ovol":"0"}
{"Name":"90srock (RR) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rr&station=90srock&bitrate=128","Port":"80","ovol":"0"}
{"Name":"alternative80s (RR) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rr&station=alternative80s&bitrate=128","Port":"80","ovol":"0"}
{"Name":"alternativerock (RR) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rr&station=alternativerock&bitrate=128","Port":"80","ovol":"0"}
{"Name":"altrock (RT) aac 128 kbps","URL":"hot.friezy.ru","File":"/?radio=rt&station=altrock&bitrate=128","Port":"80","ovol":"0"}
{"Name":"Antena 2 aac 32 kbps","URL":"live3.rcnmundo.com","File":"/antena2.aac","Port":"80","ovol":"0"}
{"Name":"Blu Musica (Pop) mp3 64 kbps","URL":"14073.live.streamtheworld.com","File":"/MUSICABLU.mp3","Port":"3690","ovol":"50"}
{"Name":"Blu Radio mp3 64 kbps","URL":"20323.live.streamtheworld.com","File":"/BLURADIO.mp3","Port":"80","ovol":"0"}
{"Name":"Cadena Dial (Pop Latino) mp3 128 kbps","URL":"20423.live.streamtheworld.com","File":"/CADENADIAL.mp3","Port":"80","ovol":"0"}

karawin, 2017-12-27:
Ok now, update and restore stations to the webradio.
Thanks

I will update KaRadio32 too soon.

arlaor, 2017-12-27:
Hi @karawin 
Excellent update
Now work AUTOPLAY too
Thanks.

karawin, 2017-12-28:
AUTOPLAY was working a while ago now.
Ok, i close it

closed 2017-12-28 by karawin

arlaor, 2017-12-28:
Hi @karawin 
Excuse me
I have noticed that AUTOPLAY only works for me with some radio stations and apparently they are the ones that are broadcast in AAC format.
When the radio station with which you must start this in MP3 format, AUTOPLAY does not work
message is 
##CLI.STOPPED# from Adc Stop

karawin, 2017-12-28:
ADC stop means that the ADC input detected a level. If you don't use a control panel, the adc must be grounded.

arlaor, 2017-12-28:
hi @karawin 
I have connected A0 to GND and I have observed the following: when I select a station in mp3 from the drop-down list and play it, without pressing the PLAY STATION button and restart the hardware, it generates the message ADC STOP.
But in another test, when I press the PLAY STATION button to play the station and after a moment I restart the hardware, the radio station starts correctly when the hardware is turned on, without generating ADC STOP message.

karawin, 2017-12-28:
Trust me, the only origin of the message is from the ADC read.
If the A0 input is grounded you must see a message in the beginning of the log with adcdiv value of 0
kprintf(PSTR("ADC Div: %d from adc: %d\n"),adcdiv,ap);
If >0 then the board is falling and the only solution is to ground the esp8266 adc pin to ground.
Some low cost boards have failing divisor resistors.

arlaor, 2017-12-28:
hi @karawin 

it is correct, of three board esp8266, only one works correctly with A0 connected to GND,
No panel
ADC Div: 0 from adc: 3
 the others show ADC DIV 3 from ADC: 11