karawin/Ka-Radio #57: Websocket

RobertSVK, 2018-02-24:
Hi, it's awesome project.

Could you please provide some description or example how to control Ka-Radio via websocket ?
I'm able to receive info via websocket after some change e.g. volume, song 

{"wsvol":"26"}
{"meta":"Bread {Bread - Retrospective} - Been Too Long On The Road"}
{"meta":"Eagles {Long Road Out Of Eden} - Busy Being Fabulous"}
{"wsvol":"17"}
{"meta":"Carole King - Where You Lead"}
{"meta":"Karen Carpenter - Remember When Lovin' Took All"}
{"monitor":"http://194.150.236.178:8000/"}
{"monitor":"http://194.150.236.178:8000/"}
{"meta":"Alessi {Big Deal (Live Without You)} - Big Deal (Live Without You)"}

But there is no effect if I try to send message.

Is it a bug or feature ? 
Or just just I don't know how to use it ?

Thanks

karawin, 2018-02-28:
The websocket interface is mainly used to send informations to all browsers connected.
Commends are sent with POST from the browser.
The only documented interface on web is the one described in http://karadio.karawin.fr/Interface.txt

karawin, 2018-09-16:
No news.
Closed but reopen if still valid

closed 2018-09-16 by karawin