karawin/Ka-Radio #59: No Sound

pilnikov, 2018-03-26:
Hello JPC. By all indications (log - ok!) must be play, but not sing. There are thoughts that the initialization of VS1053 does not pass correctly, because SPI speed is to high. Is it possible to set a low SPI speed at 0.2 MHz instead of 2 MHz? 
Thank you.
Log:
SCI_Mode (0x4800) = 0x4802
SCI_Status (0x48) = 0x4C
VS Version (VS1053 is 4) = 4
SCI_ClockF = 0xB800

karawin, 2018-05-22:
The init of the vs or any command are done in low speed and the data transfer for the song are done in higher speed according to the specs.
So i don't think it is the problem.

pilnikov, 2018-05-22:
usually it happens on "blue" boards, with "red" problems less.
Can be where that is a comparison of the circuits of these boards?

karawin, 2018-09-16:
No news.
Closed but reopen if still valid

closed 2018-09-16 by karawin