karawin/Ka-Radio #65: Ka-Radio software Bug

zsoltyfm, 2018-07-15:
Dear KaraWin!

He did a great job!
I use Wemos D1.
Sometimes the radio does not sound and the web page is unavailable.
However, this is related to wifi. In this case, the LED blinks in the same way when everything is OK. In this case, I press the reset button and everything restarts. The radio loads the radio station from the list but the volume does not load. The default remains. The volume sliders do not work properly. The website does not appear in the GUI if you change the volume from anywhere. After updating your site. The slider does not follow the value set in the machine.

zsoltyfm, 2018-07-16:
If it does not reach the radio station, it freezes. Connects to the wifi network but does not load your site either. You need to restore it to work again.

zsoltyfm, 2018-07-16:
I measured. The radio will stop every time after 36 and a half minutes. must be restored.
I'm listening to this: http://79.172.241.238:8000/musicfm.mp3

karawin, 2018-07-22:
Time to time I have a try restart in the log for this station.
Try restart means that the stream stopped to be received. 
KaraDio try to reopen the stream.

karawin, 2018-09-16:
No news.
Closed but reopen if still valid

closed 2018-09-16 by karawin