karawin/Ka-Radio #68: Move to esp-open-rtos

h1aji, 2018-11-13:
I have found a project with advanced libraries.  https://github.com/SuperHouse/esp-open-rtos

Is it possible to make Ka-radio working with it?

karawin, 2018-11-19:
May be. It is derived from the espressif esp.
But no time to test it.

h1aji, 2018-12-05:
I am trying to compile it and seeing this error:
```
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c: In function 'wifiScanCallback':
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c:281:23: error: dereferencing pointer to incomplete type
    bss_link = bss_link->next.stqe_next;
                       ^
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c:283:42: error: dereferencing pointer to incomplete type
    sprintf(buf, "\n%s;%d;%d;%d", bss_link->ssid, bss_link->channel, bss_link->rssi, bss_link->authmode);
                                          ^
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c:283:58: error: dereferencing pointer to incomplete type
    sprintf(buf, "\n%s;%d;%d;%d", bss_link->ssid, bss_link->channel, bss_link->rssi, bss_link->authmode);
                                                          ^
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c:283:77: error: dereferencing pointer to incomplete type
    sprintf(buf, "\n%s;%d;%d;%d", bss_link->ssid, bss_link->channel, bss_link->rssi, bss_link->authmode);
                                                                             ^
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c:283:93: error: dereferencing pointer to incomplete type
    sprintf(buf, "\n%s;%d;%d;%d", bss_link->ssid, bss_link->channel, bss_link->rssi, bss_link->authmode);
                                                                                             ^
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c: In function 'wifiGetStation':
/home/murad.ustarkhanov/esp/esp-open-rtos/examples/Ka-radio/interface.c:385:24: error: storage size of 'cfgg' isn't known
  struct station_config cfgg;
                        ^

```

Would you be able to assist fixing it?

karawin, 2018-12-05:
The adaptation to esp-open-rtos is not as easy as a simple compilation. You need to take into account the new include and structures of the sdk.
In this case, it seems that struct bss_info is not the same.
Idem for struct station_config

If interface.c is the first file compiled, you will find many other "problems" like these.
And as I told you: no time to help, because I don't know this sdk and don't want to load it.
Good luck.

h1aji, 2020-05-09:
Hi @karawin 
I finally managed to try to compile karadio with esp-open-rtos hoping new build will allow me to play bitrates higher than 64kbit/s

Would you be able to have a look at my repo - https://github.com/h1aji/karadio and fix the errors?
I am not very experienced with C. I would really appreciate your help.

h1aji, 2020-05-09:
Or maybe somebody else here could assist ...

pilnikov, 2020-05-09:
https://github.com/pilnikov/adf/tree/master/esp-adf/examples/Ka_mod
try........

h1aji, 2020-05-11:
@pilnikov это для карадио32?
Я справился с ошибками которые были связаны с sdk. Остаются только синтаксические. Я не программист и не могу разобраться.
Например:
> /home/hmurad/esp/karadio/src/app_main.c: In function 'user_rf_cal_sector_set':
> /home/hmurad/esp/karadio/src/app_main.c:461:5: error: unknown type name 'flash_size_map'
>      flash_size_map size_map = system_get_flash_size_map();
>      ^
> /home/hmurad/esp/karadio/src/app_main.c:461:5: warning: implicit declaration of function 'system_get_flash_size_map' [-Wimplicit-function-declaration]
> /home/hmurad/esp/karadio/src/app_main.c:463:14: error: 'FLASH_SIZE_4M_MAP_256_256' undeclared (first use in this function)
>          case FLASH_SIZE_4M_MAP_256_256:
>               ^
> /home/hmurad/esp/karadio/src/app_main.c:463:14: note: each undeclared identifier is reported only once for each function it appears in
> /home/hmurad/esp/karadio/src/app_main.c:467:14: error: 'FLASH_SIZE_8M_MAP_512_512' undeclared (first use in this function)
>          case FLASH_SIZE_8M_MAP_512_512:
>               ^
> /home/hmurad/esp/karadio/src/app_main.c:471:14: error: 'FLASH_SIZE_16M_MAP_512_512' undeclared (first use in this function)
>          case FLASH_SIZE_16M_MAP_512_512:
>               ^
> /home/hmurad/esp/karadio/src/app_main.c:472:14: error: 'FLASH_SIZE_16M_MAP_1024_1024' undeclared (first use in this function)
>          case FLASH_SIZE_16M_MAP_1024_1024:
>               ^
> /home/hmurad/esp/karadio/src/app_main.c:476:14: error: 'FLASH_SIZE_32M_MAP_512_512' undeclared (first use in this function)
>          case FLASH_SIZE_32M_MAP_512_512:
>               ^
> /home/hmurad/esp/karadio/src/app_main.c:477:14: error: 'FLASH_SIZE_32M_MAP_1024_1024' undeclared (first use in this function)
>          case FLASH_SIZE_32M_MAP_1024_1024:
>               ^
> 
> /home/hmurad/esp/karadio/src/app_main.c:575:9: error: unknown type name 'flash_size_map'
>          flash_size_map size_map = system_get_flash_size_map();
>          ^
> /home/hmurad/esp/karadio/src/upgrade_lib.c:26:7: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'struct'
>  LOCAL stru

Могли бы вы глянуть?

pilnikov, 2020-05-12:
Я свой релиз делал для карадио32. 
Для простого кардио собирал мод очень давно. Через компиляцию в линуксе. Сейчас уже не помню как. И желания нет повторять этот адов костыль. Инструментария ноль. А ковыряться текстовым редактором да еще без возможности тестовой сборки - это сами понимаете не лучшее занятие. Так что увы.

h1aji, 2020-05-25:
@pilnikov Я более менее разобрался со сборкой. Отключил ОТА, теперь все собирается только крэшится на плате. Похоже, что это тупик, так как я не разбираюсь в коде.
Кстати, можно использовать VS code.
А вообще, думается, надо портировать код от ka-radio32 на новый SDK. Я был бы готов тестировать, только кому это интересно? ...

pilnikov, 2020-05-25:
мой релиз для карадио **32** и он на новой idf **4.1**

pilnikov, 2020-05-25:
для простого карадио не очень интересно ввиду отсутствия свободных IO и ограниченностью ресурсов для подключения внешней периферии.

h1aji, 2021-06-09:
@pilnikov Недавно нашел esp-12f и собрал его с зеленой vs1053, i2c lcd 2004 и IR приемником. Все что надо на борту имеется. а дополнительную переферию можно через ардуино соединять если надо. 
Пока собрал по такой схеме и залил esp-radio, но проблема с затыками на высоких битрейтах
[image]

По хорошему karadio надо переводить на esp-open-rtos там уже реализованы вебсокет и веб сервер

h1aji, 2021-06-09:
ESP SDK 1 and 2 are dead now
@karawin I have made this project buildable in my fork here https://github.com/h1aji/karadio and would be keen to test if you continue.
Would you be able to give it a go? I mentioned how to compile in the readme, its very easy
I use VScode for editing.

closed 2023-05-10 by h1aji