karawin/Ka-Radio #77: Station edit not working

nlovrens, 2019-03-11:
Hi,

In 1.9.0 R0 it is not possible to edit stations using web GUI. Also, if all stations are deleted using "Erase all" it is not possible to add stations. Import using txt file is working. Tested with chrome and firefox.

Thank you

karawin, 2019-03-12:
Yes sorry, a little mistake.
Corrected now.
Update again please.
Thanks

nlovrens, 2019-03-12:
Hi, yes, edit of the station is working now. There is still an issue if you delete all stations, then it is not possible to add stations with the GUI because list is missing and there is no "Edit" button to push (screenshot attached).

Thanks
[image]

karawin, 2019-03-12:
Yes again. Thanks.
Ok now

closed 2019-04-10 by karawin