karawin/Ka-Radio #80: VS1053b AAC & WMA internal frequency is too high

enjoyneering, 2019-08-25:
`VS1053_WriteRegister(SPI_CLOCKF,0xB8,0x00); // SC_MULT = x1, SC_ADD= x1`

Because all VS1053 registers are 16-bit the SPI_CLOCKF nuber is 0xB800=0b1011 1000 0000 0000 where 15..13-bits is clock multiplier, 12..11-bits addition to multiplier (see p.43):

SC_MULT=1010 0000 0000 0000=0xA000= x4.0 not x1
SC_ADD=0b0001 1000 0000 0000=0x1800= x2.0 not x1

The equation for internal clock CLKI at **normal speed**: CLKI = XTALI x SC_MULT = 12.288 x 4.0 = 49.152MHz. It is OK, because the maximum VS1053 frequency is 55.3MHz (see. p7)
But when chip is trying to decode AAC & WMA it automatically choose to use addition to multiplier SC_ADD. The equation for internal clock at **boost speed**: CLKI = XTALI x (SC_MULT + SC_ADD) = 12.288 x (4.0 + 2.0)=73.728MHz with recommended maximum  55.3MHz

Recommended SC_MULT=x3.5, SC_ADD=x1 (SCI_CLOCKF=0x8800)

`VS1053_WriteRegister(SPI_CLOCKF,0xB88,0x00); // SC_MULT = x3.5, SC_ADD= x1`

Do not forget your SDI speed is CLKI / 4 = 49.152 / 4=12.288MHz & SCI speed is CLKI / 7 = 49.152 / 7=7.02MHz

karawin, 2019-08-25:
ok thanks. i will modify it.

karawin, 2019-08-25:
You mean
VS1053_WriteRegister(SPI_CLOCKF,0xB8,0x80); // SC_MULT = x3.5, SC_ADD = x1

enjoyneering, 2019-08-25:
Yes. But 0x88, 00 will work only for  12.288MHz crystal oscillator. The full equation for different crystals is more complicated & you have to change 2 registers.

I'm working on new VS1053 driver for your awesome project. Will share it some time next month.

P.S. The VS1053 can work with crystal oscillator 12-13MHz & 24-26MHz with some limitations. The recommend frequency is 12.288MHz. It gives you the highest sample rate 48KHz=12.288/256