karawin/Ka-Radio #81: Arduino ADC keyboard (red one with yellow buttons)

diver321, 2019-12-22:
Hi,
I bought the arduino ADC keyboard like this:
[image]

and only one button is working - the UP  button for changing the volume up (actually good...) so i wanna (if it is possible) to try change the settings (analog readings) in code. If it is possible where can i change those settings?

Thank You.

diver321, 2019-12-23:
Ok, i figured out by myself:
For using this red arduino keyboard you need to connect it to 3.3V pin (GND and signal to ADC pin of your choice of course) and change some things in "Adc read: keyboard buttons" section of addon.c file (before compilation):
1. in adcInit() change adc detection range - attenuation (ADC_ATTEN_DB_11) to full-scale voltage 3.9 V
2. edit adcLoop() like this:
void adcLoop() {
	uint32_t voltage,voltage0,voltage1;
	bool wasVol = false;
	if (channel == GPIO_NONE) return;  // no gpio specified
	
	voltage0 = (adc1_get_raw(channel)+adc1_get_raw(channel)+adc1_get_raw(channel)+adc1_get_raw(channel))/4;
	vTaskDelay(1);
	voltage1 = (adc1_get_raw(channel)+adc1_get_raw(channel)+adc1_get_raw(channel)+adc1_get_raw(channel))/4;
//	printf ("Volt0: %d, Volt1: %d\n",voltage0,voltage1);
	voltage = (voltage0+voltage1)*105/(819);
//	if (voltage <  40) return; // no panel  NEEDED TO COMMENT OUT FOR 5Key board
//	printf("Voltage: %d\n",voltage);

	if (inside&&(voltage0 > 3700)) 
	{
		inside = false;
		wasVol = false;
		return;
	}
	if (voltage0 > 3700) 
	{
		wasVol = false;
	}
	if ((voltage0 >3700) || (voltage1 >3700)) return; // must be two valid voltage	
	
	if (voltage < 985) ESP_LOGD(TAG,"Voltage: %i",voltage);	
		printf("VOLTAGE: %d\n",voltage);
	if ((voltage >90) && (voltage < 140)) // volume +
	{
		setRelVolume(+5);
		wasVol = true;
		ESP_LOGD(TAG,"Volume+ : %i",voltage);
	}
	else if ((voltage > 250) && (voltage < 300)) // volume -
	{
		setRelVolume(-5);
		wasVol = true;
		ESP_LOGD(TAG,"Volume- : %i",voltage);
	}	
		else if ((voltage > 420) && (voltage < 520)) // station+
		{
			if (!wasVol)
			{
				evtStation(1);
				ESP_LOGD(TAG,"station+: %i",voltage);
			}
		}	
		else if ((voltage == 0) && (voltage < 10)) // station-
		{
			if (!wasVol)
			{
				evtStation(-1);
				ESP_LOGD(TAG,"station-: %i",voltage);
			}
		}	
	if (!inside)
	{	
		if  ((voltage >2000) && (voltage < 3000)) // toggle time/info  old stop  NOT NEEDED FOR 5Key board
		{
			inside = true;
			toggletime();
			ESP_LOGD(TAG,"toggle time: %i",voltage);	
		}
		else if ((voltage >700) && (voltage < 770)) //start stop toggle   old start
		{
			inside = true;
			startStop();
			ESP_LOGD(TAG,"start stop: %i",voltage);
		}

	}
}

This worked with mine keyboard and maybe for yours will too. If not, read the values in serial monitor by pressing the keys and correct the section ranges.

karawin, 2019-12-31:
well done, but it is for KaRadio32.