karawin/Ka-Radio #86: Update to SDK 3

h1aji, 2020-05-22:
Is there any chance to get the code update to work with SDK 3?
It looks like the easiest way is to port code from Ka-Radio32 
There are lots of improvements and I am hoping to get better performance with new SDK

karawin, 2020-05-24:
No more free time for this.
I hope that someone will do it.

h1aji, 2023-05-08:
Hi JP,

I managed to port most of the code from Ka-Radio32 and the only piece I am struggling with is VS1053 driver.
Can you please help me to adjust this piece of code of yours:

```
bool VS1053_HW_init()
{
	gpio_num_t miso;
	gpio_num_t mosi;
	gpio_num_t sclk;
	gpio_num_t xcs;
	gpio_num_t xdcs;
	
	uint8_t spi_no; // the spi bus to use
	
	gpio_get_spi_bus(&spi_no,&miso,&mosi,&sclk);	
	gpio_get_vs1053(&xcs,&rst,&xdcs,&dreq);
	
	// if xcs = 0 the vs1053 is not used
	if (xcs == GPIO_NONE)
	{
		vsVersion = 0; 
		ESP_LOGE(TAG,"VS1053 not used");
		return false;
	}

	uint32_t freq = spi_cal_clock(APB_CLK_FREQ, 1600000, 128, NULL);
	ESP_LOGI(TAG,"VS1053 LFreq: %d",freq);

	spi_interface_t devcfg={
        .clock_speed_hz=freq,            //Clock out at x MHz
		.command_bits = 8,
		.address_bits = 8,
		.dummy_bits = 0,
		.duty_cycle_pos = 0,
		.cs_ena_pretrans = 0,
		.cs_ena_posttrans = 1,
		.flags = 0,	
        .mode=0,                         //SPI mode 
        .spics_io_num = xcs,             //XCS pin
        .queue_size=1,                   //We want to be able to queue x transactions at a time
        .pre_cb=NULL,  //Specify pre-transfer callback to handle D/C line
		.post_cb = NULL
	};	

	//slow speed
	ESP_ERROR_CHECK(spi_bus_add_device(spi_no, &devcfg, &vsspi));

	//high speed	
	freq = spi_cal_clock(APB_CLK_FREQ, 6100000, 128, NULL);

	ESP_LOGI(TAG,"VS1053 HFreq: %d",freq);
	devcfg.clock_speed_hz = freq;
	devcfg.spics_io_num= xdcs;               //XDCS pin
	devcfg.command_bits = 0;
	devcfg.address_bits = 0;
	ESP_ERROR_CHECK(spi_bus_add_device(spi_no, &devcfg, &hvsspi));
	
	//Initialize non-SPI GPIOs
	gpio_config_t gpio_conf;
	gpio_conf.mode = GPIO_MODE_OUTPUT;
	gpio_conf.pull_up_en = GPIO_PULLUP_DISABLE;
	gpio_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
	gpio_conf.intr_type = GPIO_INTR_DISABLE;	
	gpio_conf.pin_bit_mask = ((uint64_t)(((uint64_t)1)<<rst));
	ESP_ERROR_CHECK(gpio_config(&gpio_conf));

	gpio_conf.mode = GPIO_MODE_INPUT;
	gpio_conf.pull_up_en = GPIO_PULLUP_DISABLE;
	gpio_conf.pull_down_en = GPIO_PULLDOWN_ENABLE;
	gpio_conf.intr_type = GPIO_INTR_DISABLE;	
	gpio_conf.pin_bit_mask = ((uint64_t)(((uint64_t)1)<<dreq));
	ESP_ERROR_CHECK(gpio_config(&gpio_conf));
	
	//gpio_set_direction(dreq, GPIO_MODE_INPUT);
	//gpio_set_pull_mode(dreq, GPIO_PULLDOWN_ENABLE); //usefull for no vs1053 test
	return true;
}

```
According to https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-reference/peripherals/spi.html

Thank you

karawin, 2023-05-09:
May be refer to https://github.com/karawin/Ka-Radio ?

h1aji, 2023-05-09:
It doesnt help much tbh. I managed to fix **VS1053_spi_init**

```
void VS1053_spi_init() {
	esp_err_t ret;

	if(!vsSPI) vsSPI=xSemaphoreCreateMutex();
	if(!hsSPI) hsSPI=xSemaphoreCreateMutex();;

    ESP_LOGI(TAG, "Init GPIO");
    // Set CS and DCS pins as output and high
	gpio_reset_pin(PIN_NUM_XCS);
	gpio_set_direction(PIN_NUM_XCS, GPIO_MODE_OUTPUT);
	gpio_set_level(PIN_NUM_XCS, 1);

	gpio_reset_pin(PIN_NUM_XDCS);
	gpio_set_direction(PIN_NUM_XDCS, GPIO_MODE_OUTPUT);
	gpio_set_level(PIN_NUM_XDCS, 1);

    // Set DREQ pin as input and low
	gpio_reset_pin(PIN_NUM_DREQ);
	gpio_set_direction(PIN_NUM_DREQ, GPIO_MODE_INPUT);
	gpio_set_level(PIN_NUM_XDCS, 0);

    ESP_LOGI(TAG, "Init HSPI");

	spi_config_t spi_config={
        .interface.val = SPI_DEFAULT_INTERFACE,
        .intr_enable.val = SPI_MASTER_DEFAULT_INTR_ENABLE,
		.mode = SPI_MASTER_MODE
	};

	ret=spi_init(KSPI, &spi_config);
	assert(ret==ESP_OK);
}
```
But VS1053_HW_init doesnt seem to be very understandable

h1aji, 2023-05-10:
I need your help adjusting this piece of code. Can you please advise how it can be modified for ESP8266?

```
    spi_interface_t interface_config = {
        /* SPI mode 3, CPOL = 1, CPHA = 1 */
        .cpol = 1,
        .cpha = 1,
        .bit_tx_order = 0,
        .bit_rx_order = 0,
        .byte_tx_order = 0,
        .byte_rx_order = 0,
        .mosi_en = 1,
        .miso_en = 1,
        .cs_en = 1,
    };

	//slow speed 2MHz
	ESP_ERROR_CHECK(spi_bus_add_device(spi_no, &interface_config));

	//high speed 10MHz
	freq = spi_cal_clock(APB_CLK_FREQ, 6100000, 128, NULL);
	ESP_LOGI(TAG,"VS1053 HFreq: %d",freq);
	interface_config.clock_speed_hz = freq;
	interface_config.spics_io_num = xdcs;               //XDCS pin
	interface_config.command_bits = 0;
	interface_config.address_bits = 0;
	interface_config.
	ESP_ERROR_CHECK(spi_bus_add_device(spi_no, &interface_config));

```

h1aji, 2023-05-12:
I managed to fix Init part. Now I need to migrate read and write

```
void VS1053_spi_write_char(uint8_t *cbyte, uint16_t len) {
	esp_err_t ret;
	spi_trans_t t;
	
	memset(&t, 0, sizeof(t));       // Zero out the transaction
	t.mosi = cbyte;	
	t.bits.mosi = len * 8; 
	//t.rxlength=0;	
	while(gpio_get_level(PIN_NUM_DREQ) == 0) taskYIELD();
	spi_take_semaphore(sSPI);
	ret = spi_trans(HSPI_HOST, &t);    // Transmit!
	if (ret != ESP_OK) ESP_LOGE(TAG,"err: %d, VS1053_spi_write_char(len: %d)",ret,len);
	spi_give_semaphore(sSPI);
//	while(gpio_get_level(PIN_NUM_DREQ) == 0 );
}

void VS1053_WriteRegister(uint8_t addressbyte, uint8_t highbyte, uint8_t lowbyte) {
	spi_trans_t t;
	esp_err_t ret;

	if (vsVersion == 0) return;

	memset(&t, 0, sizeof(t));       //Zero out the transaction
	t.cmd = VS_WRITE_COMMAND;
	t.addr = addressbyte;
	t.mosi = highbyte;
	t.mosi = lowbyte;		
	t.bits.mosi = 16;
	WaitDREQ();
	spi_take_semaphore();
	ret = spi_trans(HSPI_HOST, &t);   // Transmit!
	if (ret != ESP_OK) ESP_LOGE(TAG,"err: %d, VS1053_WriteRegister(%d,%d,%d)",ret,addressbyte,highbyte,lowbyte);
	spi_give_semaphore();	
	WaitDREQ();
}

void VS1053_WriteRegister16(uint8_t addressbyte, uint16_t value) {
    spi_trans_t t;
	esp_err_t ret;

	if (vsVersion == 0) return;

	memset(&t, 0, sizeof(t));       //Zero out the transaction
	t.cmd = VS_WRITE_COMMAND;
	t.addr = addressbyte;
	t.mosi = (value>>8)&0xff;	
	t.mosi = value&0xff;		
	t.bits.mosi = 16;
	WaitDREQ();
	spi_take_semaphore();	
	ret = spi_trans(HSPI_HOST, &t);  //Transmit!	
	if (ret != ESP_OK) ESP_LOGE(TAG,"err: %d, VS1053_WriteRegister16(%d,%d)",ret,addressbyte,value);
	spi_give_semaphore();
	WaitDREQ();
}

uint16_t VS1053_ReadRegister(uint8_t addressbyte) {
	uint16_t result;
	spi_trans_t t;
	esp_err_t ret;

	if (vsVersion == 0) return 0;
	memset(&t, 0, sizeof(t));       //Zero out the transaction
	t.cmd = VS_READ_COMMAND;
	t.addr = addressbyte;
	t.bits.miso = 16;

	WaitDREQ();
	spi_take_semaphore();
	ret = spi_trans(HSPI_HOST, &t);  //Transmit!
	if (ret != ESP_OK) ESP_LOGE(TAG,"err: %d, VS1053_ReadRegister(%d), read: %d",ret,addressbyte,(uint32_t)*t.miso);
	result = (((t.mosi[0]&0xFF)<<8) | ((t.mosi[1])&0xFF));
//	ESP_LOGI(TAG,"VS1053_ReadRegister data: %d %d %d %d",t.rx_data[0],t.rx_data[1],t.rx_data[2],t.rx_data[3]);
	spi_give_semaphore();
	WaitDREQ();
	return result;
}
```
@karawin can you please have a look and point if there is anything wrong. thanks

h1aji, 2023-05-20:
The last piece that is left now is **SPIGetChar** and **SPIPutChar**
@karawin will you be able to review my code and let me know if needs correction? Thanks

```
uint8_t SPIGetChar() {
	while(READ_PERI_REG(SPI_CMD(HSPI))&SPI_USR);	//wait for SPI to be ready       

	CLEAR_PERI_REG_MASK(SPI_USER(HSPI),SPI_FLASH_MODE|SPI_USR_MOSI);
  
	SET_PERI_REG_MASK(SPI_USER(HSPI),SPI_USR_MISO);

	WRITE_PERI_REG (SPI_USER1 (HSPI),(0&SPI_USR_MOSI_BITLEN)<<SPI_USR_MOSI_BITLEN_S| //Number of bits to send
			((8-1)&SPI_USR_MISO_BITLEN)<<SPI_USR_MISO_BITLEN_S|	//Number of bits to receive
			(23&SPI_USR_ADDR_BITLEN)<<SPI_USR_ADDR_BITLEN_S);	//Number of bits in address

	SET_PERI_REG_MASK(SPI_CMD(HSPI), SPI_USR);
	while(READ_PERI_REG(SPI_CMD(HSPI))&SPI_USR) ;

	if (READ_PERI_REG(SPI_USER(HSPI))&SPI_RD_BYTE_ORDER) {
			return READ_PERI_REG(SPI_W0(HSPI))>>24;	//Assuming data in is written to MSB. TBC
	} else {
			return READ_PERI_REG(SPI_W0(HSPI));
	}
}

void SPIPutChar(uint8_t data) {
	while(READ_PERI_REG(SPI_CMD(HSPI))&SPI_USR);	//wait for SPI to be ready  

	CLEAR_PERI_REG_MASK(SPI_USER(HSPI),SPI_FLASH_MODE|SPI_USR_MISO);

    SET_PERI_REG_MASK(SPI_USER(HSPI),SPI_USR_MOSI);	//enable MOSI function in SPI module

	WRITE_PERI_REG(SPI_USER1(HSPI),(((8-1)&SPI_USR_MOSI_BITLEN)<<SPI_USR_MOSI_BITLEN_S)|   //len bits of data out
			((0&SPI_USR_MISO_BITLEN)<<SPI_USR_MISO_BITLEN_S)|       //no data in
			((23&SPI_USR_ADDR_BITLEN)<<SPI_USR_ADDR_BITLEN_S));     //address is 24 bits A0-A23

	if (READ_PERI_REG(SPI_USER(HSPI))&SPI_WR_BYTE_ORDER) {
			WRITE_PERI_REG(SPI_W0(HSPI),(uint32_t)data<<24);
	} else {
			WRITE_PERI_REG(SPI_W0(HSPI),(uint32_t)data);
	}
	SET_PERI_REG_MASK(SPI_CMD(HSPI),SPI_USR);
}
```

h1aji, 2023-05-28:
I managed to migrate the project to SDK 3.4 here - https://github.com/h1aji/Ka-Radio
But I am experiencing issues

> rst cause:4, boot mode:(3,6)

@VicVa  Any chance to have a look?

karawin, 2023-05-28:
Good job, but sorry, no free time.
It seems that your bin cannot start at all.

h1aji, 2023-05-28:
Thanks. this is why i am asking for assistance now. 
Maybe @pilnikov ?

h1aji, 2023-06-02:
I am seeing this error 

> wdt reset
> load 0x3ffe8008, len 24, room 16 
> tail 8
> chksum 0x06
> load 0x3ffe8020, len 3340, room 0 
> tail 12
> chksum 0xed
> ho 0 tail 12 room 4
> load 0x40100000, len 7216, room 12 
> tail 4
> chksum 0xd6
> csum 0xd6
> ␛[0;32mI (30) boot: ESP-IDF 3.4.0+sha.519046b 2nd stage bootloader␛[0m
> ␛[0;32mI (31) boot: compile time 01:20:56␛[0m
> ␛[0;32mI (31) boot: SPI Speed      : 40MHz␛[0m
> ␛[0;32mI (34) boot: SPI Mode       : DIO␛[0m
> ␛[0;32mI (38) boot: SPI Flash Size : 4MB␛[0m
> ␛[0;32mI (42) boot: Partition Table:␛[0m
> ␛[0;32mI (46) boot: ## Label            Usage          Type ST Offset   Length␛[0m
> ␛[0;32mI (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
> ␛[0;32mI (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
> ␛[0;32mI (68) boot:  2 factory          factory app      00 00 00010000 000f0000␛[0m
> ␛[0;32mI (75) boot:  3 ������k���Lk�␂^�*���␃ unknown          ff ff ffffffff ffffffff␛[0m
> ␛[0;32mI (83) boot: End of partition table␛[0m
> ␛[0;32mI (87) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7af0c (503564) map␛[0m
> ␛[0;32mI (96) esp_image: segment 1: paddr=0x0008af24 vaddr=0x4028af1c size=0x20b50 (133968) map␛[0m
> ␛[0;32mI (104) esp_image: segment 2: paddr=0x000aba7c vaddr=0x3ffe8000 size=0x00aa0 (  2720) load␛[0m
> ␛[0;32mI (113) esp_image: segment 3: paddr=0x000ac524 vaddr=0x40100000 size=0x00080 (   128) load␛[0m
> ␛[0;32mI (122) esp_image: segment 4: paddr=0x000ac5ac vaddr=0x40100080 size=0x057a4 ( 22436) load␛[0m
> ␛[0;32mI (131) boot: Loaded app from partition at offset 0x10000␛[0m
> ␛[0;32mI (156) Main: starting app_main()␛[0m
> ␛[0;32mI (158) Main: RAM left: 90388, Internal 0␛[0m
> ␛[0;31mE (161) Main: Running partition type 0 subtype 0 (offset 0x00010000)␛[0m
> ␛[0;31mE (165) eeprom: DEVICE Partition not found␛[0m
> ␛[0;31mE (168) eeprom: DEVICE1 Partition not found␛[0m
> ␛[0;31mE (172) eeprom: STATIONS Partition not found␛[0m
> ␛[0;31mE (177) eeprom: getDeviceSetting fails␛[0m
> ␛[0;32mI (182) Main: Partition init done...␛[0m
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x40216450  PS      : 0x00000030  A0      : 0x4021644b  A1      : 0x3ffed6d0  
> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
> A6      : 0x00000000  A7      : 0x4024cd1c  A8      : 0x00000016  A9      : 0x00000090  
> A10     : 0x4028dd62  A11     : 0x64207469  A12     : 0x4028d858  A13     : 0x00016114  
> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
> 
> Backtrace: 0x40216450:0x3ffed6d0 0x40226ab3:0x3ffed700 
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
> 

What am I missing @karawin ?

h1aji, 2023-10-15:
Hi @karawin this time I only need your help with partitions.csv
Can you tell me if this is correct configuration?

```
# Espressif ESP8266 Partition Table
# Name,   Type, SubType, Offset,   Size
nvs,      data, nvs,     0x9000,   0x4000
otadata,  data, ota,     0xd000,   0x2000
phy_init, data, phy,     0xf000,   0x1000
ota_0,    0,    ota_0,   0x10000,  0xF0000
ota_1,    0,    ota_1,   0x110000, 0xF0000
# Ka-Radio partitions
device,   0x40,      ,   ,        0x1000
stations, 0x41,      ,   ,        0x10000
device1,  0x42,      ,   ,        0x1000
hardware, data, nvs  ,   ,        0x3000
#                END =   0x400000
```

Because I see this error

> ␛[0;32mI (30) boot: ESP-IDF 3.4.0+sha.519046b 2nd stage bootloader␛[0m
> ␛[0;32mI (31) boot: compile time 23:03:30␛[0m
> ␛[0;32mI (31) boot: SPI Speed      : 40MHz␛[0m
> ␛[0;32mI (34) boot: SPI Mode       : DIO␛[0m
> ␛[0;32mI (38) boot: SPI Flash Size : 4MB␛[0m
> ␛[0;32mI (42) boot: Partition Table:␛[0m
> ␛[0;32mI (46) boot: ## Label            Usage          Type ST Offset   Length␛[0m
> ␛[0;32mI (53) boot:  0 nvs              WiFi data        01 02 00009000 00004000␛[0m
> ␛[0;32mI (60) boot:  1 otadata          OTA data         01 00 0000d000 00002000␛[0m
> ␛[0;32mI (68) boot:  2 phy_init         RF data          01 01 0000f000 00001000␛[0m
> ␛[0;32mI (75) boot:  3 ota_0            OTA app          00 10 00010000 000f0000␛[0m
> ␛[0;32mI (82) boot:  4 ota_1            OTA app          00 11 00110000 000f0000␛[0m
> ␛[0;32mI (90) boot:  5 device           unknown          40 00 00200000 00001000␛[0m
> ␛[0;32mI (97) boot:  6 stations         unknown          41 00 00201000 00010000␛[0m
> ␛[0;32mI (105) boot:  7 device1          unknown          42 00 00211000 00001000␛[0m
> ␛[0;32mI (112) boot:  8 hardware         WiFi data        01 02 00212000 00003000␛[0m
> ␛[0;32mI (119) boot:  9 �����1��)�       unknown          ff ff ffffffff ffffffff␛[0m
> ␛[0;32mI (127) boot: End of partition table␛[0m
> ␛[0;32mI (131) boot: No factory image, trying OTA 0␛[0m
> ␛[0;32mI (136) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7a878 (501880) map␛[0m
> ␛[0;32mI (145) esp_image: segment 1: paddr=0x0008a890 vaddr=0x4028a888 size=0x20a34 (133684) map␛[0m
> ␛[0;32mI (154) esp_image: segment 2: paddr=0x000ab2cc vaddr=0x3ffe8000 size=0x00aa0 (  2720) load␛[0m
> ␛[0;32mI (162) esp_image: segment 3: paddr=0x000abd74 vaddr=0x40100000 size=0x00080 (   128) load␛[0m
> ␛[0;32mI (171) esp_image: segment 4: paddr=0x000abdfc vaddr=0x40100080 size=0x057a4 ( 22436) load␛[0m
> ␛[0;32mI (180) boot: Loaded app from partition at offset 0x10000␛[0m
> ␛[0;32mI (198) Main: starting app_main()␛[0m
> ␛[0;32mI (199) Main: RAM left: 73900, Internal 0␛[0m
> ␛[0;31mE (201) Main: Running partition type 0 subtype 16 (offset 0x00010000)␛[0m
> ␛[0;31mE (204) eeprom: getDeviceSetting fails␛[0m
> ␛[0;32mI (209) Main: Partition init done...␛[0m
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x402163d2  PS      : 0x00000030  A0      : 0x402163cd  A1      : 0x3fff16d0  
> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
> A6      : 0x00000000  A7      : 0x4024c978  A8      : 0x00000016  A9      : 0x00000090  
> A10     : 0x4028d6ce  A11     : 0x64207469  A12     : 0x4028d1c4  A13     : 0x000120ac  
> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
> 
> Backtrace: 0x402163d2:0x3fff16d0 0x4022670f:0x3fff16f0 
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x3ffe8008, len 24, room 16 
> tail 8
> chksum 0x06
> load 0x3ffe8020, len 3340, room 0 
> tail 12
> chksum 0xec
> ho 0 tail 12 room 4
> load 0x40100000, len 7216, room 12 
> tail 4
> chksum 0xd7
> csum 0xd7