karawin/Ka-Radio #87: Can't connect to wifi after upgrade from 1.8R2 to 1.9R6

totalizator, 2020-08-20:
Hi,
I can't connect to wifi anymore after upgrade from 1.8R2 to 1.9R6. I have tried to reenter the AP credentials but still no go. I'm using special characters in my password but it has worked prior to the upgrade (OTA).

Any ideas?

macmattes, 2020-11-22:
hi, i have the same problem, i flashed an older version, and it worked again.
Is it possible to flash an older version via webgui?

totalizator, 2020-11-22:
Do you have multiple SSID with the same name? I suspect that the new version tries to connect to the weaker signal and fails. There is option to provide name only so I created virtual AP dedicated to Karadio and this sort of solved it.

Still I consider this as a bug. Any chances for a fix (support for BSSID)?

macmattes, 2021-02-06:
hi yes, i have Network with many AP's with the same SSID. Dedicated virtual AP's are not an option, because it contradicts the centralized management of the AP's. I can't know when and where i want to use the radio in the area.