karawin/Ka-Radio #89: Unable to see the SSID network

ChristopheBaeyens, 2020-12-09:
I'm searching for days now in the hope of getting this to work.
But after loading the BIN files I discovered no new network.
Checked the web tried some other combinations but had no luck.
Started from this site https://www.instructables.com/Wifi-Webradio-With-ESP8266-and-VS1053/
But finally used the loader as mentioned in the readme here but still no progress.
[image]

Just, checked my esp8266 with a simple led test via the intranet and it is still working.

ChristopheBaeyens, 2020-12-09:
I just rebooted due to a windows update, and this I did not expect, I can see the SSID now, but for the moment still no sound...

ChristopheBaeyens, 2020-12-09:
Unable to connect to the site anymore. Flashed the board / Erased it / Redownloaded all bin's, The SSID is comming up but unable to connect to the site http://192.164.4.1/ anymore....
For the moment also removed all cables. But nothing I do seems to get the site back up...
Does someone has any sugestions?

pilnikov, 2020-12-10:
>  unable to connect to the site http://192.16**4**.4.1/ anymore....
> Does someone has any sugestions?

You must connect to 192.16**8**.4.1

ChristopheBaeyens, 2020-12-10:
Ok must have been tired yesterday, thx for your feedback.
Had again some problems with my one of my laptops USB connections, strange updating the board works with 1 usb port but to get the board working I had to switch to another USB port.
I finally got the project working, but still have a buffering problem.
I also do not understand why A0 has to be connected to the ground (i did not do it for the moment)

ChristopheBaeyens, 2020-12-10:
Did some search, and I suppose I need extra ram for buffering. Would it be possible to get some references of external ramboards, if possible without the need of soldering?

pilnikov, 2020-12-11:
i know chip only, no ramboards

ChristopheBaeyens, 2020-12-11:
Hi Alex,

 

How did you solve the buffer problem, or did you only use those radio’s without this type of problem?

 

Christophe.

pilnikov, 2020-12-11:
i use esp32, but i have some problem even with 4mb ext ram

pilnikov, 2020-12-11:
W (18910) webclient: Stream found!!!! Start player
##CLI.PLAYING#
##CLI.META#: Nina - The reason is you
I (22660) vs1053: Wait while Buffer fill 0%
I (22710) vs1053: Wait while Buffer fill 0%
I (22760) vs1053: Wait while Buffer fill 0%
I (22810) vs1053: Wait while Buffer fill 0%
I (22860) vs1053: Wait while Buffer fill 0%
I (22910) vs1053: Wait while Buffer fill 0%
I (22960) vs1053: Wait while Buffer fill 0%
I (23010) vs1053: Wait while Buffer fill 0%
I (23060) vs1053: Wait while Buffer fill 0%
I (23110) vs1053: Wait while Buffer fill 0%
I (23160) vs1053: Wait while Buffer fill 1%
I (23210) vs1053: Wait while Buffer fill 1%
I (23260) vs1053: Wait while Buffer fill 1%
I (23310) vs1053: Wait while Buffer fill 1%
I (23360) vs1053: Wait while Buffer fill 1%
I (23410) vs1053: Wait while Buffer fill 1%
I (23460) vs1053: Wait while Buffer fill 1%
I (23510) vs1053: Wait while Buffer fill 1%
I (23560) vs1053: Wait while Buffer fill 2%
I (23610) vs1053: Wait while Buffer fill 2%
I (23660) vs1053: Wait while Buffer fill 2%
I (23710) vs1053: Wait while Buffer fill 2%
I (23760) vs1053: Wait while Buffer fill 2%
I (23810) vs1053: Wait while Buffer fill 2%
I (23860) vs1053: Wait while Buffer fill 2%
I (23910) vs1053: Wait while Buffer fill 2%
I (23960) vs1053: Wait while Buffer fill 2%
I (24010) vs1053: Wait while Buffer fill 2%
I (24060) vs1053: Wait while Buffer fill 2%
I (24110) vs1053: Wait while Buffer fill 2%
I (24160) vs1053: Wait while Buffer fill 3%
I (24210) vs1053: Wait while Buffer fill 3%
I (24260) vs1053: Wait while Buffer fill 3%
I (24310) vs1053: Wait while Buffer fill 3%
I (24360) vs1053: Wait while Buffer fill 3%
I (24410) vs1053: Wait while Buffer fill 3%
I (24460) vs1053: Wait while Buffer fill 3%
I (24510) vs1053: Wait while Buffer fill 3%
I (24560) vs1053: Wait while Buffer fill 3%
I (24610) vs1053: Wait while Buffer fill 4%
I (24660) vs1053: Wait while Buffer fill 4%
I (24710) vs1053: Wait while Buffer fill 4%
I (24760) vs1053: Wait while Buffer fill 4%
I (24810) vs1053: Wait while Buffer fill 4%
I (24860) vs1053: Wait while Buffer fill 4%
I (24910) vs1053: Wait while Buffer fill 4%
I (24960) vs1053: Wait while Buffer fill 4%
I (25010) vs1053: Wait while Buffer fill 5%
I (25080) vs1053: Wait while Buffer fill 4%
I (25130) vs1053: Wait while Buffer fill 4%
I (25180) vs1053: Wait while Buffer fill 5%
I (25310) vs1053: Wait while Buffer fill 4%
I (25360) vs1053: Wait while Buffer fill 4%
I (25410) vs1053: Wait while Buffer fill 5%
I (25540) vs1053: Wait while Buffer fill 4%
I (25590) vs1053: Wait while Buffer fill 4%
I (25640) vs1053: Wait while Buffer fill 4%
I (25690) vs1053: Wait while Buffer fill 4%
I (25740) vs1053: Wait while Buffer fill 4%
I (25790) vs1053: Wait while Buffer fill 5%
I (25950) vs1053: Wait while Buffer fill 4%
I (26000) vs1053: Wait while Buffer fill 5%
I (39430) audio_player: Buffer fill 9%, 102867 // 1048576 bytes
I (60290) audio_player: Buffer fill 9%, 102738 // 1048576 bytes
I (81100) audio_player: Buffer fill 9%, 104253 // 1048576 bytes
I (102370) audio_player: Buffer fill 9%, 98733 // 1048576 bytes
I (123200) audio_player: Buffer fill 9%, 98588 // 1048576 bytes
I (143450) audio_player: Buffer fill 10%, 106303 // 1048576 bytes
I (164350) audio_player: Buffer fill 10%, 106158 // 1048576 bytes
I (185420) audio_player: Buffer fill 9%, 102941 // 1048576 bytes

my log right now

ChristopheBaeyens, 2020-12-11:
Ok, in that case I will close the case this evening.

Thx.

By the way do you know why A0 should be grounded?

pilnikov, 2020-12-11:
to avoid interference that can be recognized by the esp as pressing the buttons of an analog keyboard

pilnikov, 2020-12-11:
Can I use the Arduino ide monitor for this?
Yes of course

ChristopheBaeyens, 2020-12-11:
OK, something I will do this evening then, thx again 😊

ChristopheBaeyens, 2020-12-11:
For the moment I will wait until I have decided to go for another board or extra ram.

closed 2020-12-11 by ChristopheBaeyens