karawin/Ka-Radio #90: Password protected WifiKaradio AP or a configurable delay and/or tries count to connect a stored network

andykarpov, 2021-02-02:
Is it possible to add WEP / WPA encryption for the WifiKaradio network or add a configurable delay to start the internal WifiKaradio network after a configurable amount of tries ?

Case that scaring me a lot:
1) My KaRadio starts much faster that the router (after power failure / outage, for example)
2) In this case KaRadio is unable to connect the stored wifi network and starts it's own WifiKaradio network (widely open)
3) Anybody can connect that WifiKaradio and see a stored password in the settings

karawin, 2021-02-05:
You are right. 
The password must not be clear in the message.
I will study it.

andykarpov, 2021-02-14:
Sweet, i guess this issue is already fixed by v2.1 R0 and can be closed

karawin, 2021-02-15:
Yes, it is.
I close. Thanks for reporting the problem.

closed 2021-02-15 by karawin