karawin/Ka-Radio #92: Is it possible to change the address of the time server?

cyzzar, 2021-03-05:
Please add the ability to change the address for the NTP server. The default one is not always available.

karawin, 2021-03-06:
No. But why? no problem with the default one here since 3 years now.
Some ponctual ntp fails on logs but no more than 1 per day.

cyzzar, 2021-03-09:
The fact is that I assembled an add-on for a 20x4 display and as additional information I display the current time on it.

I poll the time with the sys.date command twice a minute. But quite often the request returns an error. 
There are two options to fix this. 1 - add a clock module to the arduino. 2 - change the address of the NTP server to the local one. 

The second option is preferable. 
It is a pity if this is impossible to implement.