karawin/Ka-Radio #95: Flashing

Zinci, 2021-03-20:
Could you please edit the line on the main page:
user1.4096.new.4.bin at 0x1000 
to
user1.4096.new.4.bin at 0x10000

It took me quite a few sleepless nights to work this out and maybe it will help someone like me (a total newbie).
I received by mistake couple of ESP8266 and I did not want to throw them away. I guess I will have a kitchen Internet Radio 
BTW the ESP32 works perfectly. :+1: 
I am in the process of putting them inside the iPod docking station, hopefully. :fearful:

Thanks for the code!

molotovec, 2021-03-22:
Have no idea how did you manage it working. I've tried a couple of boards with params from main page and have no luck - board isn't available in the wifi list and no blinking after flash  button[used ESP download tool v3.6.4]
boot_v1.6.bin at 0x0000,
user1.4096.new.4.bin at 0x10000, **(0x01000 - from instructables)**
user2.4096.new.4.bin at 0X81000,
esp_init_data_default.bin at 0x3FC000
and blank.bin at 0x3fe000

> Could you please edit the line on the main page:
> user1.4096.new.4.bin at 0x1000
> to
> user1.4096.new.4.bin at 0x10000
> 
> It took me quite a few sleepless nights to work this out and maybe it will help someone like me (a total newbie).
> I received by mistake couple of ESP8266 and I did not want to throw them away. I guess I will have a kitchen Internet Radio
> BTW the ESP32 works perfectly. 👍
> I am in the process of putting them inside the iPod docking station, hopefully. 😨
> 
> Thanks for the code!

Zinci, 2021-03-22:
After your post I was worried that my success was a fluke, so I successfully flashed **another** ESP8266 with the corrected address (0X10000). I even found a screenshot from flash-tool here:
https://github.com/karawin/Ka-Radio/blob/master/ESP8266-Firmware/documents/karadiov365.jpg
I downloaded the "fresh" binaries into **_bin_** directory though.

Zinci, 2021-03-29:
Just remembered something. Did you reset manually the board after flashing? As far as I know the flash tool does not reset the board after upload.

> Have no idea how did you manage it working. I've tried a couple of boards with params from main page and have no luck - board isn't available in the wifi list and no blinking after flash button[used [...]

Zinci, 2021-03-30:
Thanks, I just tried it and it does work. I do not know what happened when I tried previously. 
Apart from the device name in Wireless Network Watcher which is a bit weird and comes up as ESP_294C60.home but I can access it from WifiKaRadio.home.

closed 2021-03-30 by Zinci